MarlinFirmware/Marlin #24164: [BUG] (failed Marlin on Mac OS)

jihyungchun, 2022-05-13:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I've download the Marlin-bug-fix-2.0.x and the Configurations-bug-fix-2.0 for the BTT SKR min E3 V2. I've made my edits to the configurations and configurations.adv.h and saved. I've went to auto build Marlin and clicked the hammer. I got the following options

STM32F103RC_btt
STM32F103RC_btt_USB
STM32F103RE_btt
STM32F103RE_btt_USB
STM32F103RC_btt_maple
STM32F103RC_btt_USB_maple
STM32F103RE_btt_maple
STM32F103RE_btt_USB_maple

I am not sure which one to choose? but I chose STM32F103RC_btt and clicked Build but sometimes it's successful and sometimes not. It just says failed and no other info is given. Even when it is sucessful, I can't tell where the firmware I've built went to

### Bug Timeline

new

### Expected behavior

I expected it to build and provide me a firmware

### Actual behavior

Keeps failing to build 

### Version of Marlin Firmware

bugfix-2.0.x

### Printer model

Ender 3 Pro

### Electronics

BTT SKR mini E3 V2.0

### Add-ons

CR touch

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

thisiskeithb, 2022-05-13:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2022-05-13 by thisiskeithb

jihyungchun, 2022-05-13:
Configuration_adv.h error
#include errors detected. Please update your includePath. Squiggles are disabled for this translation unit (/Users/jihyungchun/Downloads/Marlin-bugfix-2.0.x/Marlin/src/MarlinCore.cpp).
Configuration.h error
#include errors detected. Please update your includePath. Squiggles are disabled for this translation unit (/Users/jihyungchun/Downloads/Marlin-bugfix-2.0.x/Marlin/src/MarlinCore.cpp).