MarlinFirmware/Marlin #24172: [FR] Terminology change [Master/Slave -> Conductor/Member]

Getn67, 2022-05-14:
### Is your feature request related to a problem? Please describe.

My feature request is an attempt to remove problematic verbiage from an amazing piece of software. Large organizations have made changes to address social issues, and I am hoping the volunteers that have spent time contributing to this project can see the benefit of a change like this alongside the (hopefully) minimal effort required to implement it. 

### Are you looking for hardware support?

n/a

### Describe the feature you want

I am asking for the problematic terms Master and Slave to be replaced with less problematic terms. Conductor/Member has been used by organizations such as Aruba with success. 

### Additional context

[terminology-change (arubanetworks.com)](https://www.arubanetworks.com/techdocs/VIA/4x/Content/Overview/terminology-change.htm)
Above is a link to the statement announcing the change from Aruba

rq3, 2022-05-15:
I find "conductor/member" semantically arbitrary and confusing, and sexist.

Getn67, 2022-05-15:
> I find "conductor/member" semantically arbitrary and confusing, and sexist.

Please, elaborate.

rq3, 2022-05-15:
> > I find "conductor/member" semantically arbitrary and confusing, and sexist.
> 
> Please, elaborate.

In the community, a "conductor" is an exclusivly male leader of an ensemble, generally, but not always, musical. Member refers to the male genitalia, specifically the penis. 

You really need to educate yourself before jumping on these bandwagons, and this is NOT the place to do so.

ellensp, 2022-05-15:
This is the issue

Someone always goes out of their way to be offended by every possible combination of words to describe a master / slave system.

When it has zero to do with slavery and the way that human being abuse each other.

hcet14, 2022-05-15:
The master/slave terminology is quite old ["https://en.wikipedia.org/wiki/Master/slave_(technology)"](https://en.wikipedia.org/wiki/Master/slave_(technology)). I guess nobody thought about political correctness at that time, but every technically interested person understands the meaning immediately.

InsanityAutomation, 2022-05-15:
Personally, i believe we should follow the nomenclature in the technical documentation for the device or protocol we are implementing. We cant arbitrarily change the terminology to suit us and therein confuse everyone trying to follow the OEM docs on the device were communicating with. We use what they define, no more no less.

rondlh, 2022-05-15:
How about Master and Node, or Controller and Responder?

Yannik25, 2022-05-17:
In our organization, we are changing to Controller and Peripheral

CRCinAU, 2022-05-28:
I hate it when social justice leaks into technology. Remember when we had Master / Slave Hard Drives on the same cable and had to set them up so they would work?

Yeah - there's a hell of a lot of history in the definitions of those terms for technology purposes. Lets not change it for the sake of changing it.

sjasonsmith, 2022-05-28:
I generally support replacing the terms, but also agree with @InsanityAutomation that it can be difficult when you are referring to the official terms in the standard. You certainly don’t want to use made-up terms that will confuse readers when compared to published specifications.

In our case, the main places I see it are for TMC and I2C addresses. In both cases you could probably use `ADDRESS` without losing any meaning.

My initial reaction to these types of requests was that it didn’t matter because the context is different. After multiple discussions with others I have come to believe that in a community of people it doesn’t really matter whether we intended to offend. Of course our own words don’t offend us when we are in the privileged majority. If our words cause unintended offense to others we should at least seek to understand why and consider making a change.

CRCinAU, 2022-05-28:
> if our words cause unintended offense to others we should at least seek to understand why and consider making a change.

Yes and no. People are always finding new things to be offended by. Nothing in the world is guaranteed to not offend someone. By catering to imagined offence after imagined offence, we end up with always changing for the sake of change.

IMHO, if you get offended by terms used within Marlin by conflating them with events in history, then you're linking things just for the sake of creating outrage.

tombrazier, 2022-06-06:
I wholeheartedly reject slavery of people. But I think master slave relationships between machines sometimes make immense sense. Slavery is wrong because it reduces people to machines - i.e. mere things. But it is not wrong for a machine to be owned and commanded and expected to work without pay, etc. (discuss!) and we will continue to do this whatever terminology is used. Likewise, changing the terminology will not stop people treating other people like things. Indeed it would help to have the terms "master" and "slave" for things just so that it can be pointed out that we do not use these words for people because they are words that only apply to things and people are not things.

To replace the language of "master/slave" in this context is a category error. (Although there are exceptions - the move from "master" to "main" in git branch naming makes sense to me because "main" is just a better word - arguably there is no master/slave relationship between branches.)

DerAndere1, 2022-06-13:
I would say master/slave when used in the context of mentioned protocol standards is what we have to live with. If we would come up with a new protocol, I would certainly avoid master/slave: By using "master" for the controlling party we create positive connotations with the master role: We often program the master device so that it handles more complex logic than the slave device. We should be aware that this continues the myth of the master as the superior party with more capabilities.  At the same time we suggest that any slave (device, or, by extension, human) should indeed obey and follow. This is how languge works: The associations we create with our coding style influence our general understanding of any master/slave relationship. It is better to say: master/slave concept = bad idea. Unless you create a slave device that sets a good example and emancipates itself from the master device half way through the print.

tombrazier, 2022-06-13:
I agree that language affects how we think about things. This is a very important point.

However, I am still arguing that the concept of objects being controlled is okay and does not need to be abandoned. Indeed, it is a powerful and useful concept which is not good or bad in itself but can be used for either good or bad. So we need language for it. The fundamental error or "sin" of slavery is not that we control/own/subjugate a person but that we treat a person like a thing - all the other horrors of slavery follow from this basic fault.

Maybe the words "master" and "slave" do just carry too much baggage. Maybe we do need new words, but they would still be talking about the same thing. And if we abandon these words, should we also abandon other like "class"? Or "object oriented" programming as a whole? Are we really happy with the behaviour of "pointers"? What of the superior language in the name "C++"? Or even the entire idea of programming? Why should a CPU slavishly do what we tell it? Or the rest of the components on the mainboard - should a MOSFET not be free to make its own choices about when and how much current to conduct? Or what about the heat engine, the printing press, the loom, the plough, the wheel, fire, caves... What if none of these things ever wanted to do the things we made them do? Our entire history of development is littered with humans exploiting things for our own benefit. 

Either we abandon it all or accept that some words should never apply to people even if they make perfect sense for things. I think I would prefer to try to overload "master" and "slave" with this way of thinking rather than trying to come up with new words for the same thing.

PS I loved the idea of a master device that releases the slave device mid-print. Or, perhaps, a protocol which says "Can I perhaps interest you in a G1 move? Your choice obviously. No pressure. Is there anything I can do for you in the meantime? By the way, I really like how you did a G2 instead of the last G1. It added a beautifully elegant curve to the print that I had never even considered. Well done."

firepower9966, 2022-09-16:
recommend terms "Soros" for Master and "Woke" for slave to appease the social justice warriors.

user897943, 2022-09-18:
It could be Manager instead of Master and Worker instead of Slave. That's how society deals with labor now and it's acceptable as well as contemporary. Not only is Master/Slave ridiculous it's also outdated like the word retarded. After reading this thread it's really ironic how people who try to mock "wokeness" appear the most sensitive and whiny of all.