MarlinFirmware/Marlin #24177: Some PIO updates and warnings.  LPC1769 (BTT E3 turbo)

badbod, 2022-05-15:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

In file included from Marlin\src\sd\../inc/MarlinConfigPre.h:37,
                 from Marlin\src\sd\../inc/MarlinConfig.h:28,
                 from Marlin\src\sd\cardreader.cpp:23:
Marlin\src\sd\cardreader.cpp: In constructor 'CardReader::CardReader()':
Marlin\src\sd\../inc/../core/macros.h:279:49: warning: 'void* memset(void*, int, size_t)' clearing an object of non-trivial type 'class SdFile'; use assignment or value-initialization instead [-Wclass-memaccess]
  279 | #define ZERO(a)             memset(a,0,sizeof(a))
      |                                                 ^
Marlin\src\sd\cardreader.cpp:172:3: note: in expansion of macro 'ZERO'
  172 |   ZERO(workDirParents);
      |   ^~~~
In file included from Marlin\src\sd\cardreader.h:49,
                 from Marlin\src\sd\cardreader.cpp:29:
Marlin\src\sd\SdFile.h:41:7: note: 'class SdFile' declared here
   41 | class SdFile : public SdBaseFile {

### Bug Timeline

New, started 5 or 6 days ago after PIO updates

### Expected behavior

Normally only warning is about assigning Z2 to E1 which is indicative of correct and helpful behaviour.

### Actual behavior

unexpected and slightly alarming warning issued.

### Steps to Reproduce

Just Build,
 warning can change some depending on Alpha sort options (one combo causes build fail altogether) but always generates this or similar related warning regardless

### Version of Marlin Firmware

2.0.9.3

### Printer model

Ender 3 V2

### Electronics

BTT SKR E3 Turbo

### Add-ons

Dual Z steppers

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Additional information & file uploads

[E3-Turbo-Marlin-2.0.9.3.zip](https://github.com/MarlinFirmware/Marlin/files/8694737/E3-Turbo-Marlin-2.0.9.3.zip)

oliver-eifler, 2022-05-15:
Thought it was already fixed

fast fix - to satisfy pointer type casting and memaccess warning
in /src/core/macros.h
change (line 221)
`#define ZERO(a)             memset(a,0,sizeof(a))`
to
`#define ZERO(a)             memset((void*)a,0,sizeof(a))`

works for my e3 turbo fine since > 1 month, seemed to have no negative effect even with stm32 & avr cpu.

badbod, 2022-05-15:
Hi @oliver-eifler  Thanks,   It is still in bugfix as of 5 mins ago, but your quick fix did the trick.   :-)

thisiskeithb, 2022-05-16:
Closing since a PR was opened:

- https://github.com/MarlinFirmware/Marlin/pull/24178

closed 2022-05-16 by thisiskeithb