MarlinFirmware/Marlin #24184: [BUG] MKS Tinybee / ESP32 MKS12864 display does not work because of u8g_esp32_spi.cpp

jesterhead82, 2022-05-16:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

[BUG] MKS Tinybee / ESP32 MKS12864 display does not work because of u8g_esp32_spi.cpp

MKS fork for the Tinybee version of it:

#if ENABLED(MKS_MINI_12864_V3)
  #define MDOGLCD_MOSI                     23
  #define MDOGLCD_SCK                      18
  #define MLCD_RESET_PIN                   0
  #define MLCD_PINS_DC                     4
  #define MDOGLCD_CS                       21
  #define MDOGLCD_A0                       4
#endif

#if ENABLED(MKS_MINI_12864) <---12864 triggers that one
  #define MDOGLCD_MOSI                     23
  #define MDOGLCD_SCK                      18
  #define MLCD_RESET_PIN                   -1
  #define MLCD_PINS_DC                     15
  #define MDOGLCD_CS                       16
  #define MDOGLCD_A0                       15
#endif

Recent Marlin Versions overwrite the Pins simply

### Bug Timeline

At least in recent versions

### Version of Marlin Firmware

2.0.xx

ellensp, 2022-05-16:
You did NOT try bugfix
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/HAL/ESP32/u8g_esp32_spi.cpp#L28

ellensp, 2022-05-16:
right this was updated 2 months ago...

ellensp, 2022-05-16:
Please test this PR

jesterhead82, 2022-05-16:
Hi,

can confirm that at least that issue is now gone with this TinyBee(ast) when I apply your PR.

ellensp, 2022-05-17:
Since PR has been created to fix this, closing this issue.

closed 2022-05-17 by ellensp