MarlinFirmware/Marlin #24185: [BUG] `G28` accidentally set active extruder to 0

RedSaltyFish, 2022-05-16:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

`G28` accidentally set active extruder to 0. If brim is meant to be printed by extruder 1, Cura will put `T1` at the very first line of the generated gcode file. Performing auto home afterwards will accidentally select extruder 0, and make the first layer printed by the wrong nozzle.

Gcode snippet:
```
;FLAVOR:Marlin
;TIME:25804
;Filament used: 9.1121m, 4.85233m
;Layer height: 0.2
;MINX:95.251
;MINY:70.953
;MINZ:0.3
;MAXX:260.998
;MAXY:171.282
;MAXZ:60.9
;Generated with Cura_SteamEngine 4.13.1
T1 ;Select extruder 1
M140 S60
M105
M190 S60
M104 T0 S190
M104 S230
M105
M109 T0 S190
M105
M109 S230
M82 ;absolute extrusion mode
G28 ;Home, and accidentally select extruder 0
```

Terminal output:
[image]

### Bug Timeline

I upgraded from Marlin 1.1.9 to 2.0.x yesterday. Don't know exactly when the issue started with Marlin 2.0.x.

### Expected behavior

`G28` won't accidentally select extruder 0.

### Actual behavior

`G28` accidentally select extruder 0.

### Steps to Reproduce

1. Import a model that needs two extruders;
2. Select the second extruder to print the brim;
3. Print, and observe which extruder is extruding.

### Version of Marlin Firmware

Marlin bugfix-2.0.x

### Printer model

Homemade printer with ATmega2560 controller board

### Electronics

ATmega2560 controller board

### Add-ons

None

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Additional information & file uploads

Log (probably not very useful):
```
Send: T1
Recv: echo:T1
Recv: Active Extruder: 1
Recv: ok
[...]
Send: G28
Recv: echo:G28
Recv: Active Extruder: 0
```

[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/8701632/Configurations.zip)

RedSaltyFish, 2022-05-16:
I can see it is intentional in `G28.cpp`. But this behavior is not compatible with Cura when the brim should be printed by the second extruder.

RedSaltyFish, 2022-05-16:
This is the gcode I plan to print:
[gcode.zip](https://github.com/MarlinFirmware/Marlin/files/8702621/gcode.zip)

These are the logs before the wrong extruder began to extrude:
[logs.zip](https://github.com/MarlinFirmware/Marlin/files/8702689/logs.zip)

`log.zip` contains 2 log files. `log_full.txt` is the full log, and `log_suppressed.txt` is the log with temperature and processing responses suppressed.

Looking further into the logs, it seems like `G28` was not causing the problem, but the `G29` between 2 `G28`s.

InsanityAutomation, 2022-05-16:
I see multiple return paths in G29 without reverting the changed tool, though most of them come from an error path. I think the tool change is called too early, and should be done after some of the error checking but before movement. By the look of it, it should work if it gets to where it prints the topography report as there is no further return statements. My guess is somewhere abl.tool_index is getting reset, but id need to do some more digging.

closed 2022-07-27 by github-actions[bot]