MarlinFirmware/Marlin #24190: [BUG] Incorrect pinout of MKS Monster8 EXP1 and EXP2

rondlh, 2022-05-16:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

The MKS Monster8 V1.0 EXP1 and EXP2 pinout is incorrect. From pins_MKS_MONSTER8.h:
```
/**
 *                  ------                                      ------
 *   (BEEPER) PB2  |10  9 | PE10 (BTN_ENC)    (SPI1 MISO) PA6  |10  9 | PA5 (SPI1 SCK)
 *   (LCD_EN) PE11 | 8  7 | PD10 (LCD_RS)       (BTN_EN1) PE9  | 8  7 | PA4 (SPI1 CS)
 *   (LCD_D4) PD9    6  5 | PD8  (LCD_D5)       (BTN_EN2) PE8    6  5 | PA7 (SPI1 MOSI)
 *   (LCD_D6) PE15 | 4  3 | PE7  (LCD_D7)       (SPI1_RS) PB11 | 4  3 | RESET
 *             GND | 2  1 | 5V                             GND | 2  1 | 3.3V
 *                  ------                                      ------
 *                   EXP1                                        EXP2
 */
```

But actually the Expansion connector notches are on the right side, and Pin 1 (square pad) is at the top left, see attached picture:
[image]
[image]

Here the official MKS documentation: https://github.com/makerbase-mks/MKS-Monster8/tree/main/hardware

Note: MKS uses a different convention for the notches than most other suppliers, which will cause issues when using the Expansion ports with for example the Full graphics display. The easiest solution is to remove the MKS expansion connector shells and flip them. They can be pulled of easily (careful!) and put back the other way around.

I believe the correct pinout is like this:
```
/**
 *                  ------                                      ------
 *   (BEEPER) PB2  | 1  2 | PE10 (BTN_ENC)    (SPI1 MISO) PA6  | 1  2 | PA5 (SPI1 SCK)
 *   (LCD_EN) PE11 | 3  4 | PD10 (LCD_RS)       (BTN_EN2) PE9  | 3  4 | PA4 (SPI1 CS)
 *   (LCD_D4) PD9  | 5  6   PD8  (LCD_D5)       (BTN_EN1) PE8  | 5  6   PA7 (SPI1 MOSI)
 *   (LCD_D6) PE15 | 7  8 | PE7  (LCD_D7)     (SD_DETECT) PB11 | 7  8 | RESET
 *            GND  | 9 10 | 5V                            GND  | 9 10 | 3.3V
 *                  ------                                      ------
 *                   EXP1                                        EXP2
 */
```
I noticed this issue is also present in various other MKS boards, like for example the pins_MKS_SGEN_L and pins_MKS_SGEN_L_V2.

### Bug Timeline

Don't know

### Expected behavior

Correct pinout

### Actual behavior

Pinout doesn't reflect actual situation

### Steps to Reproduce

Check official data vs. definition in pins file.

### Version of Marlin Firmware

Marlin 2.0.9.3 latest bugfix

### Printer model

Custom

### Electronics

MKS Monster8 V1.0

### Add-ons

BTT TFT35 V3.0

thisiskeithb, 2022-05-16:
This has been explained to the maintainer and he wants it done this way for all MKS boards (all MKS boards are “backwards”).

closed 2022-05-16 by thisiskeithb

rondlh, 2022-05-17:
Thanks for your feedback!
I expected that already, but it's clearly wrong. In my view at least a comment about this issue should be put in the pins-file.
I hooked up the Full Graphic display incorrectly because of this issue. It took me quite some time to figure out where I went wrong.

thisiskeithb, 2022-05-17:
> In my view at least a comment about this issue should be put in the pins-file.

We have one here: 

https://github.com/MarlinFirmware/Marlin/blob/0b787e8ed3d1743728eff28ccbb247cbb3101f04/Marlin/Configuration.h#L2484