MarlinFirmware/Marlin #24196: Disable MIN/MAX pin for 4th axis

Fusseldieb, 2022-05-17:
### Description

If I enable a 4th "linear_axis" in Marlin, it wants a MIN or MAX pin to home it. However, since I use the forth axis as a **rotary** one, it technically hasn't a MIN or MAX.

### Steps to Reproduce

1. Uncomment LINEAR_AXIS and set it to 4
2. Set HOME_DIR and other stuff that makes sense
3. Try to compile, it compains about the missing MIN/MAX PLUG

**Expected behavior:** I should be able to define a 4th axis without the need to specify a MIN or MAX pin. Maybe a `I_NO_PLUG` would be nice.

**Actual behavior:** Explained above

#### Additional Information

No additional information

DerAndere1, 2022-05-18:
For safety reasons, we assume that all axes, also rotary axes, have limits. If you want an infinite 4th axis, you cannot home that axis, so it does not make sense to define `I_HOME_DIR`.  If you want an infinite rotary axis, you currently have to disable the following options in Configuration.h: USE_*_MIN_PLUG, *_HOME_DIR, MIN_SOFTWARE_ENDSTOP_*, MAX_SOFTWARE_ENDSTOP_*  (replace "\*" with the internal axis ID). E.g., for the 4th axis, verify that you have the following settings:
```
//USE_I_MIN_PLUG
//I_HOME_DIR -1
//MIN_SOFTWARE_ENDSTOP_I
//MAX_SOFTWARE_ENDSTOP_I
```

In Addition, you have to remove some sanity checks in file SanityCheck.h. E.g., for the 4th axis in current bugfix-2.0.x:
https://github.com/MarlinFirmware/Marlin/blob/c814fe98d7e250692908af03836a7c362a5ce843/Marlin/src/inc/SanityCheck.h#L1481-L1489
change that to 
```
#if HAS_I_AXIS
  #if HAS_I_ENABLE && !defined(I_ENABLE_ON)
    #error "I_ENABLE_ON is required for your I driver with NUM_AXES >= 4."
  #endif
#endif
```

DerAndere1, 2022-05-18:
Yeah I just tested it and Marlin indeed expects at least one limit switch per axis.

DerAndere1, 2022-05-18:
Related: https://github.com/MarlinFirmware/Marlin/issues/18348 and https://github.com/MarlinFirmware/Marlin/issues/23281

DerAndere1, 2022-05-18:
I was able to compile with these settings: 
```
//#define USE_I_MIN_PLUG
//#define USE_I_MAX_PLUG
#define I_HOME_DIR 0
//#define MIN_SOFTWARE_ENDSTOP_I
//#define MAX_SOFTWARE_ENDSTOP_I
``` 

If you want to skip movement of the 4th axis during G28 and mark it as homed, change G28.cpp:
https://github.com/MarlinFirmware/Marlin/blob/c814fe98d7e250692908af03836a7c362a5ce843/Marlin/src/gcode/calibrate/G28.cpp#L480
change it to:
```
        if (doI) 
          #if NONE(USE_IMIN_PLUG, USE_IMAX_PLUG)
            set_axis_is_at_home(I_AXIS),
          #else 
            homeaxis(I_AXIS),
          #endif
```

Fusseldieb, 2022-07-18:
Do not stale

fiveorders, 2022-08-22:
> For safety reasons, we assume that all axes, also rotary axes, have limits.
> 
> EDIT: the approach below does not work because Marlin expects at least one endstop per axis!!!
> 
> If you want an infinite 4th axis, you cannot home that axis, so it does not make sense to define `I_HOME_DIR`. If you want an infinite rotary axis, you currently have to disable the following options in Configuration.h: `USE_*_MIN_PLUG`,`USE_*_MAX_PLUG`, `*_HOME_DIR`, `MIN_SOFTWARE_ENDSTOP_*`, `MAX_SOFTWARE_ENDSTOP_*` (replace `*` with the internal axis ID). E.g., for the 4th axis, verify that you have the following settings:
> 
> ```
> //#define USE_I_MIN_PLUG
> //#define USE_I_MAX_PLUG
> //#define I_HOME_DIR -1
> //#define MIN_SOFTWARE_ENDSTOP_I
> //#define MAX_SOFTWARE_ENDSTOP_I
> ```
> 
> In Addition, you have to remove some sanity checks in file SanityCheck.h. E.g., for the 4th axis in current bugfix-2.0.x:
> 
> https://github.com/MarlinFirmware/Marlin/blob/c814fe98d7e250692908af03836a7c362a5ce843/Marlin/src/inc/SanityCheck.h#L1481-L1489
> 
> 
> change that to
> ```
> #if HAS_I_AXIS
>   #if HAS_I_ENABLE && !defined(I_ENABLE_ON)
>     #error "I_ENABLE_ON is required for your I driver with NUM_AXES >= 4."
>   #endif
> #endif
> ```

This has been changed, I think. You need to disable the error at 2418 now. 

`    #elif LINEAR_AXES >= 4 && I_HOME_TO_MIN && DISABLED(USE_IMIN_PLUG)
     //  #error "Enable USE_IMIN_PLUG when homing I to MIN." `

Fusseldieb, 2022-10-25:
Do not stale

closed 2023-01-05 by github-actions[bot]