MarlinFirmware/Marlin #24200: Compiling of Configuration.h giving me a FAILED message

ChrisFix13, 2022-05-17:
###Did you test the latest bugfix-2.0.x code?

Yes, I'm actually working with that code only.

### Description

 I am trying to setup my Big Tree Tech SKR V1.4 for my new Creality Ender 3 Pro, so I'm following a step by step guide using a YT video. I did all of the changes needed prior to compiling the Configuration.h file and it should be an easy/quick thing to do. For some reason I get errors and the process ends. So The compiling fails.

### Steps to Reproduce

1. Change the files from platformio.ini, Configuration.h, and Configuration_adv.h like in the video here (https://www.youtube.com/watch?v=-Gdk0wHg51w&ab_channel=TeachingTech). The process starts at minute 8:26 and the problem I have is at minute 10:47. 
2.  When I compile my files I get the following in the terminal:

> Executing task in folder Marlin-bugfix-2.0.x: C:\Users\Christian Jensen\.platformio\penv\Scripts\platformio.exe run <

Warning! `src_filter` configuration option in section [env:include_tree] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:DUE] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:esp32] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_build_flags` configuration option in section [env:linux_native] is deprecated and will be removed in the next release! Please use `build_src_flags` instead      
Warning! `src_filter` configuration option in section [env:linux_native] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_build_flags` configuration option in section [env:simulator_windows] is deprecated and will be removed in the next release! Please use `build_src_flags` instead 
Warning! `src_filter` configuration option in section [env:SAMD51_grandcentral_m4] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:malyan_M300] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:STM32F103CB_malyan] is deprecated and will be removed in the next release! Please use `build_src_filter` instead    
Warning! `src_filter` configuration option in section [env:teensy31] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:teensy35] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:teensy36] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:teensy41] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Processing LPC1768 (board: nxp_lpc1768; platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.3.zip; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nxplpc-arduino-lpc176x/nxp_lpc1768.html
PLATFORM: NXP Arduino LPC176x (0.1.3) > NXP LPC1768
HARDWARE: LPC1768 100MHz, 31.97KB RAM, 464KB Flash
DEBUG: Current (cmsis-dap) On-board (cmsis-dap) External (blackmagic, jlink)
PACKAGES:
 - framework-arduino-lpc176x @ 0.2.9
 - toolchain-gccarmnoneeabi @ 1.100301.220327 (10.3.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ strict
Found 2 compatible libraries
Scanning dependencies...
Dependency Graph
|-- Servo @ 1.0.0
|-- U8glib-HAL @ 0.5.2
Building in release mode
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\HAL.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\HAL_SPI.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\MarlinSerial.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\MinSerial.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\eeprom_flash.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\eeprom_sdcard.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\eeprom_wired.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\fast_pwm.cpp.o
In file included from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:50,
                 from Marlin\src\HAL\LPC1768\HAL.cpp:24:
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:1467:4: error: #error "Leveling in Marlin requires three or more axes, with Z as the vertical axis."
 1467 |   #error "Leveling in Marlin requires three or more axes, with Z as the vertical axis."
      |    ^~~~~
In file included from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:50,
                 from Marlin\src\HAL\LPC1768\HAL_SPI.cpp:51:
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:1467:4: error: #error "Leveling in Marlin requires three or more axes, with Z as the vertical axis."
 1467 |   #error "Leveling in Marlin requires three or more axes, with Z as the vertical axis."
      |    ^~~~~
In file included from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:50,
                 from Marlin\src\HAL\LPC1768\MarlinSerial.cpp:26:
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:1467:4: error: #error "Leveling in Marlin requires three or more axes, with Z as the vertical axis."
 1467 |   #error "Leveling in Marlin requires three or more axes, with Z as the vertical axis."
      |    ^~~~~
In file included from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:50,
[... 211 lines cut ...]
Marlin\src\HAL\LPC1768\../../inc/../core/types.h:463:69: error: 'y' was not declared in this scope
  463 |   FI void set(const XYval<T> pxy)                      { x = pxy.x; y = pxy.y; }
      |                                                                     ^
compilation terminated due to -fmax-errors=5.
*** [.pio\build\LPC1768\src\src\HAL\LPC1768\eeprom_wired.cpp.o] Error 1
====================================================================================================== [FAILED] Took 5.62 seconds ======================================================================================================

Environment    Status    Duration
-------------  --------  ------------
LPC1768        FAILED    00:00:05.624
================================================================================================= 1 failed, 0 succeeded in 00:00:05.624 =================================================================================================The terminal process "C:\Users\Christian Jensen\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

3. I don't know what to do, don't want to mess with the other files so please help. 

**Expected behavior:** [What you expect to happen]
Expected it to compile fine and give me access to the bin file.

**Actual behavior:** [What actually happens]
Gave me a FAIL message and 5 Problems, 4 from the types.h file and 1 from SanityCheck.h file

#### Additional Information

[Error_Files.zip](https://github.com/MarlinFirmware/Marlin/files/8711984/Error_Files.zip)
[image]
[image]

eromaredo, 2022-05-18:
This is actually issue with a Platform.IO, not Marlin. Here is an issue https://github.com/platformio/platformio-core/issues/4270 .

ellensp, 2022-05-18:
Error message is clear #error "Leveling in Marlin requires three or more axes, with Z as the vertical axis."
You have a bad Config files

You have not enabled any stepper drivers
So marlin think your don't have any axes

You need to define your stepper drivers eg

#define X_DRIVER_TYPE  TMC2209
#define Y_DRIVER_TYPE  TMC2209
#define Z_DRIVER_TYPE  TMC2209
#define E0_DRIVER_TYPE TMC2209

ellensp, 2022-05-18:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/qNnSzsug
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2022-05-18 by ellensp

ChrisFix13, 2022-05-18:
Thank you so much @ellensp, I appreciate the help and you were right! Also, thanks for the other message, I will check them if ever I have another problem!