MarlinFirmware/Marlin #24206: Unable to compile any version of Marlin

jigneshsinha, 2022-05-18:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

thisiskeithb, 2022-05-18:
You'll need to update your PlatformIO Core.

In VSCode, click the PIO alien/ant icon & then Upgrade PlatformIO Core 

[image]

jigneshsinha, 2022-05-18:
wow, that was the most efficient response. Thank you for the help, that fixed the issue i have spent days trying to solve. Much appreciated!

thisiskeithb, 2022-05-18:
I'm going to keep this open for visibility.

LemarinelNet, 2022-05-18:
Sooo easy to fix after trying to toggle verbose for an hour, not seeing any error and finding this issue ;)

akinilicali, 2022-05-20:
[image]
I have exactly the same problem which hasn't gone away. I tried many pio and python versions. I briefly got it to work using an older Python version but now, it doesn't work with that version either.

Bob-the-Kuhn, 2022-05-21:
The PlatformIO core must be at 6.0.1 or higher for VScode to compile Marlin properly.  

Here's what mine looked like after the upgrade:

[image]

I wasn't aware of the **Upgrade PlatformIO Core** option detailed above so I typed the following into the command line prompt at the bottom of the terminal window:
**C:\Python38\python.exe -m pip install  platformio**  

I've put a red circle around the two areas of interest.

akinilicali, 2022-05-21:
Well, I tried all the cores: 6.01, older, development, etc. Fiddled with Python versions going back and forward. I had zero problems earlier on but for the last few days, I haven't been able to compile any robin nano v3 firmware. I reinstalled Vscode 4 times resetting all the folders etc.

ellensp, 2022-05-22:
@akinilicali your error has nothing to do with this issue. 
you have enabled SDIO_SUPPORT when your controller does not support SDIO

akinilicali, 2022-05-22:
> @akinilicali your error has nothing to do with this issue. you have enabled SDIO_SUPPORT when your controller does not support SDIO

Well, thank you indeed. You saved my day. I absent-mindedly enabled the SDIO_SUPPORT mistaking it for SDCARD support and disregarded the warning messages in yellow thinking they would not prevent the success of the compiling process. Silly me.

closed 2022-06-01 by thisiskeithb