MarlinFirmware/Marlin #24208: [FR] Smart dual cooling fan control for IDEX system, don't run fans that are not in use

rondlh, 2022-05-18:
I'm building an IDEX printer, both tools have their own independently controllable hotend and cooling fans (separate PWM IO pins for all 4 fans). 
I found that REDUNDANT_PART_COOLING_FAN can run both cooling fans in sync, but I hope it could be done a bit smarter by not running the cooling fan of the tool that is not in use. This would reduce the noise when the system when the system is not in DUPLICATION mode.

**IDEX FULL_CONTROL or AUTO_PARK mode:**
If T0 is active then the T0 cooling fan runs based on M106 (FAN_PIN), T1 cooling fan doesn't run.
If T1 is active then the T1 cooling fan runs based on M106 (Use REDUNDANT_PART_COOLING_FAN index). T0 cooling fan doesn't run.

**IDEX DUPLICATION mode:**
T0 and T1 cooling fans run in sync. (the way it is now when using REDUNDANT_PART_COOLING_FAN).

Doing it this way would make the process transparent for the slicer, no additional gcode would be needed.
The hotend fans (E0_AUTO_FAN_PIN, E1_AUTO_FAN_PIN) also work in this way, so it would be nice if the part cooling fans could do the same.

rondlh, 2022-05-20:
I've had a look at the code and found some odd things I don't understand.
In M106_M107.cpp
```
#if ENABLED(SINGLENOZZLE)
  #define _ALT_P active_extruder
  #define _CNT_P EXTRUDERS
#else
  #define _ALT_P _MIN(active_extruder, FAN_COUNT - 1)
  #define _CNT_P FAN_COUNT
#endif
```
Both M106 and M107 start like this:
```
  const uint8_t pfan = parser.byteval('P', _ALT_P);
  if (pfan >= _CNT_P) return;
  #if REDUNDANT_PART_COOLING_FAN
    if (pfan == REDUNDANT_PART_COOLING_FAN) return;
  #endif
```
The _ALT_P macro sets the default value for the P-parameter to 0 for tool 0 (T0) and to 1 for tool 1 (T1). REDUNDANT_PART_COOLING_FAN is 1 for my IDEX setup.
This means that M106 and M107 do nothing if tool 1 is selected and no P-parameter or P1 is provided. I do not understand this logic.
Either way, I implemented the cooling fan logic as described, testing is in progress...

thisiskeithb, 2022-05-20:
`REDUNDANT_PART_COOLING_FAN` wasn’t added for IDEX. It simply allows you to run two part cooling off two separate pins so you don’t overload a single (usually smaller) fan mosfet.

- https://github.com/MarlinFirmware/Marlin/pull/21888

rondlh, 2023-09-02:
Seems nobody is interested in this

closed 2023-09-02 by rondlh