MarlinFirmware/Marlin #24217: [BUG] (No Printer Attached no matter what serial # selected)

Sonicboom247, 2022-05-20:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

with 3/31 Bugfix i can compile fine for an SKR2 RevB with an MKS TFT28 with the below serial settings
#define SERIAL_PORT 1
#define BAUDRATE 250000
#define SERIAL_PORT_2 2
#define SERIAL_PORT_3 -1

with Bugfix i downloaded around 12:30 p,m. EST with the same settings i get no printer attached.  have tested a multitude of ports which i have on a spreadsheet and no matter what i get no printer attached,  I go back to the firmware from 3/31 and I am good.

### Bug Timeline

started today with new Bugfix 2.0

### Expected behavior

serial port 1 to detect printer

### Actual behavior

No printer attached no matter what port is defined

### Version of Marlin Firmware

02010000 from this afternoon

### Printer model

Artillery Sidewinder X1

### Electronics

Stock display on BTT SKR2 with 2209 steppers

### Add-ons

BLTouch

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Additional information & file uploads

[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/8744049/Configurations.zip)

ellensp, 2022-05-21:
You probably plugged the tft cable in to the SWD port to the left of the TFT port

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/qNnSzsug
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

Sonicboom247, 2022-05-21:
sorry your assumption is wrong, as stated i can switch to an earlier version with no issue.  I didn't state anything about moving cables, this is not a new hardware install and has been working fine. but whatever, i can figure it out.

Sonicboom247, 2022-05-21:
Thank you for reopening, seems like the version i downloaded minutes before this change caused the issue, i just grabbed a fresh version and it is working now [🔧](https://github.com/MarlinFirmware/Marlin/commit/2bb9e0d58e6513b47f90e729876de98ac7ccdade) [Remove LCD_SERIAL_PORT defaults, warn on auto-assign (](https://github.com/MarlinFirmware/Marlin/commit/2bb9e0d58e6513b47f90e729876de98ac7ccdade)https://github.com/MarlinFirmware/Marlin/pull/24170[)](https://github.com/MarlinFirmware/Marlin/commit/2bb9e0d58e6513b47f90e729876de98ac7ccdade)

closed 2022-05-21 by Sonicboom247