MarlinFirmware/Marlin #24219: [BUG] Homing issue

AfroPsycho, 2022-05-21:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Recently my motherboard is fried (BTT SKR 2.0 F429) while printing after receiving the new board (BTT SKR 2.0 **F407** ) I installed it today and it shows a tmc connection error 
I found out that two of my stepper driver (BTT TMC2209) is reporting all low 
I have tried to change stepper driver around and confirmed that two stepper drivers are not working I have unplugged them and tried to home all axis 
Z lowered a bit as usual X and Y make buzzing sound moved just a little bit and printer halted
I tried it again and same result
Than I tried to home the axis individually Z is fine but x and y is broken
Then I plugged x stepper motor to the z plug on motherboard tried to home the z axis (x motor) and it moved fine 
Then plugged the z motor to the x plug tried to home it made buzzing sound moved only little bit (possibly because the vibration) and printer halted

So something with marlin or mainboard is making that buzzing sound and I don't know what that is

### Bug Timeline

after replacing the F429 mainboard with f407

### Version of Marlin Firmware

2.0.x

### Printer model

3D4E 4GenPro 2R

### Electronics

Upgraded to SKR2 with TMC2209

### Bed Leveling

ABL Linear grid

### Your Slicer

Cura

### Host Software

OctoPrint

rondlh, 2022-05-21:
Check your board pinout for the steppers and the stepper jumpers (if any) under the drivers.

ellensp, 2022-05-22:
This is not a user support forum. This is very unlikely to be a Marlin Bug, way more likely to be a hardware issue or miss configuration. 

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/qNnSzsug
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose). forum.

closed 2022-05-22 by ellensp