MarlinFirmware/Marlin #24224: [BUG] - Host prompt does not include option to continue after Advanced pause initiated by runout sensor.

beebenutz, 2022-05-21:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

The requested host prompt for "paused" action command does not include an option to "continue". The host prompt options are "PurgeMore" or "DisableRunout". Selecting "DisableRunout" allows continuation without RunOut protection. 

### Bug Timeline

Not sure. 

### Expected behavior

Host prompt for action command "paused" , should include option to "continue"

### Actual behavior

Host prompt for action command "paused" has only options to "purgeMore" or "DisableRunout".

### Steps to Reproduce

- Begin print.
- Initiate Filament Runout by removing material from filament sensor.
- Machine notifys of filament runout on T0.
- Machine parks Nozzle and notifys of nozzle parked.
- Insert material in to filament sensor.
- Select "continue" on "Nozzle Parked" Action Command.
- Action Command "Paused" now shows.
- "Paused" action command has two options, "PurgeMore" or "DisableRunout" 
- PS I currently have Pullup and pulldown disabled for filament runout sensor. I have tried enabling each as well.

### Version of Marlin Firmware

2.0.9.3

### Printer model

Was an Anycubic Vyper

### Electronics

BTT SKR2, BTT TFT35 V3.0, Aftermarket Mechanical filament runout switch

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Simplify3D

### Additional information & file uploads

Configs: https://drive.google.com/file/d/1WNhxfUfObu6EbhsjGYAeE9GalLqeFh4I/view?usp=sharing

Couldn't get logging to work. 

Thanks!

InsanityAutomation, 2022-05-22:
That's the typical behavior when the runout sensor is still tripped when the dialog is called. If you insert the filament a little late, try to purge more once and see if it detect the reset, as it verifies on extruder motion.

beebenutz, 2022-05-22:
You have provided the missing bit of information I required to solve the issue. Maybe the code can be repaired to allow my desired configuration. 
I initially was attempting to use Marlin 2.0 official release. 
I found that Marlin Official had a bug where the Automatic unloading of the filament could not be configured within the ADVANCED_PAUSE_FEATURE. The filament would ALWAYS be unloaded upon parking of the extruder.
I do not want to have filament automatically unload upon PARK due to the filament being pushed backward through my filament sensor which would cause damage to the machine. I would prefer to unload/load manually. 
I set ALL filament travel distances under ADVANCED_PAUSE_FEATURE to ZERO in an attempt to stop the UNLOAD process. 
I found that the UNLOAD always occured and no changes to the distance parameters had any affect. 
I then transferred all my configs to the latest BUGFIX.
BUGFIX fixed the auto UNLOAD issue I was initially concerned with.
My settings under ADVANCED_PAUSE_FEATURE now affected the actions of ADVANCED_PAUSE_FEATURE.
Since my ADVANCED_PAUSE_FEATURE setting were ALL ZERO for extruder travel distances, AND the filament sensor is only reset with extruder action, my machine was stuck in a loop. 
As a work-around, I adjusted  ADVANCED_PAUSE_PURGE_LENGTH under ADVANCED_PAUSE_FEATURE to a positive value.
I now get the "continue" option, and the print now resumes normally, but with a small unwanted PURGE_LENGTH
Can the filament sensor not be reset via another function to allow MANUAL feeding of the extruder during parking?

Thank you for your knowledge regarding extruder motion and filament sensor resets!
Without that tiny bit of info, I could have never figured it out.

closed 2022-08-02 by github-actions[bot]