MarlinFirmware/Marlin #24235: Ender 3 Pro with 4.2.2 board can not compile

tomha2014, 2022-05-23:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I am not able to build marlin for an Ender 3 Pro with a 4.2.2 board and a STM32F103RET6 chip
I have testing this in the lastest stable build and nightly build.
Steps to reproduce:

Instead I get this:
Error: Build environment 'STM32F103RET6_creality' is incompatible with BOARD_CREALITY_V4. Use one of these: STM32F103RE_creality, STM32F103RE_creality_xfer, STM32F103RC_creality, STM32F103RC_creality_xfer, STM32F103RE_creality_maple

I saw somewhere to look in the pins.h folder for CREALITY_V4 and see what environment I have:
#include "stm32f1/pins_CREALITY_V4.h"         // STM32F1                                env:STM32F103RE_creality env:STM32F103RE_creality_xfer env:STM32F103RC_creality env:STM32F103RC_creality_xfer env:STM32F103RE_creality_maple

I do not have those chips I have the STM32F103RET6

I read trying to flash wrong firmware will brick the printer, rather not do that!!!!

Is the combination of toys STM32F103RET6/4.2.2 on Ender 3 Pro not supported?

Also saw you can build the Ender 3 V2 and set the display flag to be CM10(?) and it will work. Again brick concern

Also over Christmas I added a CR Touch to the Ender3 Pro and adjusted the firmware for it and it worked great.
So did something change?

### Bug Timeline

yesterday. 

### Expected behavior

the firmware to compile with out issues.

### Actual behavior

Processing STM32F103RET6_creality (platform: ststm32; board: genericSTM32F103RE; framework: arduino)
--------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Error: Build environment 'STM32F103RET6_creality' is incompatible with BOARD_CREALITY_V4. Use one of these: STM32F103RE_creality, STM32F103RE_creality_xfer, STM32F103RC_creality, STM32F103RC_creality_xfer, STM32F103RE_creality_maple
===== [FAILED] Took 0.41 seconds =====

### Steps to Reproduce

1. Download the latest Marlin 2.0.9.3 and Configuration files to match
2. unzip both files
3. from the Configurations-release-2.0.9.3/config/examples/Creality/Ender-3 Pro/CrealityV422 copy the files and place them in the Marlin-2.0.x/Marlin folder REPLACING the existing files
4. open Visual Studio code with PlatformIO installed.
5. open the project using platformIO open command
6. Go into platformIO.ini and change the default env to default_envs = STM32F103RET6_creality to match the chip
7. click on the check mark and should build and be happy

### Version of Marlin Firmware

2.0.9.3 latest

### Printer model

Creality Ender 3 Pro

### Electronics

stock

### Add-ons

CR Touch

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Slic3r

### Host Software

OctoPrint

### Additional information & file uploads

At this point I am trying to get a successful build before I change any printer stuff. So no CR touch etc.

tomha2014, 2022-05-23:
BTW I watched ever video I could find, Looks like it worked 3 months ago using Creality/ Ender 3 Pro V1.5 configuration files.

ellensp, 2022-05-23:
Please read the error

`Error: Build environment 'STM32F103RET6_creality' is incompatible with BOARD_CREALITY_V4. 
`
Right? so it does not like STM32F103RET6_creality

`Use one of these: STM32F103RE_creality, STM32F103RE_creality, STM32F103RC_creality, STM32F103RC_creality_xfer, STM32F103RE_creality_maple`

So it is expecting one of these

STM32F103RE_creality
STM32F103RE_creality_xfer
STM32F103RC_creality
STM32F103RC_creality_xfer

So use one of these.

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/qNnSzsug
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2022-05-23 by ellensp

tomha2014, 2022-05-23:
Just assuming you can use different chip environment, isn't that like opening your computer and seeing an Intel I7 chip and saying I want to build for M1 because I hear it's faster. 
Is there some document somewhere that says
"if you have "STM32F103RET6" stamped on your main chip can can use one of these"

BTW: I totally believe you!
Thank You 
Tom