MarlinFirmware/Marlin #24237: Interruption filament runout sensor doesn't work in real time and do not check itself after press "continue/resume"

svd05, 2022-05-24:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

Interruption filament runout sensor doesn't work in real time and do not check itself after press "continue/resume"

### Expected behavior

1. After triggered sensor all moves stop immediately and head parking in set coordinates.
2. After press button "continue/resume" marlin check status runout pins and if its triggered, than its will be still in parking (filament not loaded)

### Actual behavior

1. After the sensor is triggered, the head continues to move from 5 to 20 seconds. After that, he parks for a change of plastic.
Its happens when firmware configure with interruptions on main board and interruptions in display.
2. Filament not loaded, but after press "continue\resume" printing continue and sensor do not react for anything.

### Version of Marlin Firmware

2.0.9.3

### Printer model

custom

### Electronics

BTT SKR Pro v1.2

### Add-ons

max31865, I2C_EEPROM

### Bed Leveling

No Bed Leveling

### Your Slicer

Cura

### Additional information & file uploads

[config.zip](https://github.com/MarlinFirmware/Marlin/files/8766258/config.zip)

thinkyhead, 2022-05-28:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

svd05, 2022-05-30:
Ok. I trying, but on today I have another error when compile 2.0.x (2.0.9.3 + skr pro v1.2 + max31865 don't have this problem): 
[image]

svd05, 2022-08-08:
problem doesn't not fixed in last firmware version.

closed 2022-10-20 by github-actions[bot]