MarlinFirmware/Marlin #24240: [BUG] menu_probe_offset.cpp build error

wykdtwisted, 2022-05-25:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I am trying to compile 2.0.x with VS for a FLSUN QQS-Pro.  I have attempted the build using bugfix and the associated up-to-date config files.  

**During the build I get the following error:**

[image]

**The if statement of code is as follows:**

  
    if ((FINE_MANUAL_MOVE) > 0.0f && (FINE_MANUAL_MOVE) < 0.1f) {
    char tmp[20], numstr[10];
    // Determine digits needed right of decimal
    const uint8_t digs = !UNEAR_ZERO((FINE_MANUAL_MOVE) * 1000 - int((FINE_MANUAL_MOVE) * 1000)) ? 4 :
                         !UNEAR_ZERO((FINE_MANUAL_MOVE) *  100 - int((FINE_MANUAL_MOVE) *  100)) ? 3 : 2;
    sprintf_P(tmp, GET_TEXT(MSG_MOVE_N_MM), dtostrf(FINE_MANUAL_MOVE, 1, digs, numstr));
    #if DISABLED(HAS_GRAPHICAL_TFT)
      SUBMENU_F(FPSTR(NUL_STR), []{ _goto_manual_move_z(float(FINE_MANUAL_MOVE)); });
      MENU_ITEM_ADDON_START(0 + ENABLED(HAS_MARLINUI_HD44780));
      lcd_put_u8str(tmp);
      MENU_ITEM_ADDON_END();
    #else
      SUBMENU_F(tmp, []{ _goto_manual_move_z(float(FINE_MANUAL_MOVE)); });
    #endif
  }

### Bug Timeline

New, started on build

### Expected behavior

Unknown

### Actual behavior

Unknown

### Version of Marlin Firmware

2.0.x Bugfix

### Printer model

FLSUN QQS-Pro

### Electronics

Stock

### Add-ons

None

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Additional information & file uploads

[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/8771177/Configuration_adv.zip)
[menu_probe_offset.zip](https://github.com/MarlinFirmware/Marlin/files/8771201/menu_probe_offset.zip)

wykdtwisted, 2022-05-25:
My apologize for any improper formatting errors with this request.  I attempted to follow all the guidelines.  I am not a programmer, and I am a mildly computer literate (but I'm trying).

ellensp, 2022-05-25:
Issue was introduced https://github.com/MarlinFirmware/Marlin/commit/a5e1d4c50ae40c8203d8490be83ef86e16ebf1c2

tpruvot, 2022-05-25:
could be grouped with #24239 so...

tpruvot, 2022-05-25:
This should be fixed soon, seems recent changes not finished...

ellensp, 2022-05-25:
These are us working on the problem...  be patient
 
find
`  SUBMENU_F(tmp, []{ _goto_manual_move_z(float(FINE_MANUAL_MOVE)); });`
replace with
`  SUBMENU_F(FPSTR(tmp), []{ _goto_manual_move_z(float(FINE_MANUAL_MOVE)); });`

Looks like it should work

tpruvot, 2022-05-25:
or just F(tmp)

wykdtwisted, 2022-05-25:
FANTASTIC!!  Thank you all so much I appreciate the help!  Amazing

ellensp, 2022-05-26:
@tpruvot can you add this to your PR https://github.com/MarlinFirmware/Marlin/pull/24239

tpruvot, 2022-05-26:
right, i can

ellensp, 2022-05-28:
Since a pr has been created, closing this issue

closed 2022-05-28 by ellensp