MarlinFirmware/Marlin #24243: [BUG] Steinhart-Hart C coefficient is absent in configuration_adv.h being computed in temperature.cpp

Nikolay-Po, 2022-05-26:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

The problem is Steinhart-Hart C coefficient is missing in **configuration_adv.h** thermistor declarations and is missing in **reset_user_thermistors** method. So, Marlin user can't use Steinhart-Hart C coefficient to improve sensor tolerance.

Method **user_thermistor_to_deg_c** of **Temperature** class **temperature.cpp** source, computes temperature by raw ADC value using user-supplied sensor parameters. Along with resistances and Beta coefficient, **user_thermistor_to_deg_c** method accounts Steinhart-Hart C coefficient.

### Bug Timeline

This is an old issue, it persists from an introduction of user_thermistor_to_deg_c

### Expected behavior

In **configuration_adv.h**:
```//
// Custom Thermistor 1000 parameters
//
#if TEMP_SENSOR_0 == 1000
        #define HOTEND0_PULLUP_RESISTOR_OHMS         4700.0 // Pullup resistor
        #define HOTEND0_RESISTANCE_25C_OHMS        106429.8 // Resistance at 25C
        #define HOTEND0_BETA                       5067.155 // Beta value
        #define HOTEND0_SH_C_COEFF              1.84160E-07 // Steinhart-Hart C coefficient
#endif
```
In **temperature.cpp**:
```void Temperature::reset_user_thermistors() {
user_thermistor_t default_user_thermistor[USER_THERMISTORS] = {
        #if TEMP_SENSOR_0_IS_CUSTOM
                { true, HOTEND0_SH_C_COEFF, 0, HOTEND0_PULLUP_RESISTOR_OHMS, HOTEND0_RESISTANCE_25C_OHMS, 0, 0, HOTEND0_BETA, 0 },
        #endif
```
Note the definition of HOTEND0_SH_C_COEFF above.

User can define "C" coefficient of Steinhart-Hart equation.
[image]

### Actual behavior

In **configuration_adv.h**:
```//
// Custom Thermistor 1000 parameters
//
#if TEMP_SENSOR_0 == 1000
        #define HOTEND0_PULLUP_RESISTOR_OHMS         4700.0 // Pullup resistor
        #define HOTEND0_RESISTANCE_25C_OHMS        145816.4 // Resistance at 25C
        #define HOTEND0_BETA                       4371.899 // Beta value
#endif
```
Definition of "C" coefficient is missing. So, user can't setup this part of equation.
Also in **reset_user_thermistors** method "C" coefficient is unconditionally initialized by zero.
User can't tune thermistor parameters for better fit. The tolerance of temperature sensor is degraded. See "best fit" without of "C" coefficient below and compare with upper.
[image]

### Steps to Reproduce

It is enough to open configuration_adv.h file and look at "Custom Thermistor 1000 parameters" section. "C" coefficient is missing in definitions.
Also look at **Temperature::reset_user_thermistors()** method implementation. Second parameter of **user_thermistor_t** type is forced to zero instead to be initialized by user supplied value.

### Version of Marlin Firmware

Marlin bugfix-2.0.x (May 26 2022 11:20:51)

### Printer model

Anycubic Mega X

### Electronics
[... 143 lines cut ...]
{ true, HOTEND7_SH_C_COEFF, 0, HOTEND7_PULLUP_RESISTOR_OHMS, HOTEND7_RESISTANCE_25C_OHMS, 0, 0, HOTEND7_BETA, 0 },
#endif
#if TEMP_SENSOR_BED_IS_CUSTOM
{ true, BED_SH_C_COEFF, 0, BED_PULLUP_RESISTOR_OHMS, BED_RESISTANCE_25C_OHMS, 0, 0, BED_BETA, 0 },
#endif
#if TEMP_SENSOR_CHAMBER_IS_CUSTOM
{ true, CHAMBER_SH_C_COEFF, 0, CHAMBER_PULLUP_RESISTOR_OHMS, CHAMBER_RESISTANCE_25C_OHMS, 0, 0, CHAMBER_BETA, 0 },
#endif
#if TEMP_SENSOR_COOLER_IS_CUSTOM
{ true, COOLER_SH_C_COEFF, 0, COOLER_PULLUP_RESISTOR_OHMS, COOLER_RESISTANCE_25C_OHMS, 0, 0, COOLER_BETA, 0 },
#endif
#if TEMP_SENSOR_PROBE_IS_CUSTOM
{ true, PROBE_SH_C_COEFF, 0, PROBE_PULLUP_RESISTOR_OHMS, PROBE_RESISTANCE_25C_OHMS, 0, 0, PROBE_BETA, 0 },
#endif
#if TEMP_SENSOR_BOARD_IS_CUSTOM
{ true, 0, 0, BOARD_PULLUP_RESISTOR_OHMS, BOARD_RESISTANCE_25C_OHMS, 0, 0, BOARD_BETA, 0 },
#endif
#if TEMP_SENSOR_REDUNDANT_IS_CUSTOM
{ true, REDUNDANT_SH_C_COEFF, 0, REDUNDANT_PULLUP_RESISTOR_OHMS, REDUNDANT_RESISTANCE_25C_OHMS, 0, 0, REDUNDANT_BETA, 0 },
#endif
};
COPY(user_thermistor, default_user_thermistor);
}
```
Hope this help someone.

Bob-the-Kuhn, 2022-05-26:
You're correct - **sh_c_coeff** is already part of the temperature calculation.

Googling **steinhart-hart coefficients calculator** pops up multiple useable webpages.  Maybe a pointer to one of them should be added to the comments in configuration_adv.h.

Nikolay-Po, 2022-05-27:
@Bob-the-Kuhn, the problem is not in Steinhart-Hart coefficients understanding, all mathematics in Marlin is correct.
The problem is no way to use one of the coefficients, "C" by user configuration.
In my examples above the difference of using/not using "C" coefficient is clear. I'm doing numeric fitting of coefficients using complete Steinhart-Hart equation. So, I need to be able to program "C" coefficient in configuration_adv.h along with Series resistance, 25 degrees resistance and Beta coefficient.
A limited actions are required to close this ticket: just add Steinhart-Hart C coefficient into custom thermistor templates of configuration_adv.h and add these definitions into reset_user_thermistors method of temperature.cpp source. After that, anybody who needed more precise temperature control, will be able to use "C" coefficient to improve thermistor Steinhart-Hart model.
Thank You.

Nikolay-Po, 2022-06-29:
How can I help to modify temperature.cpp and configuration_adv.h? Please, point me to instruction

Nikolay-Po, 2022-06-29:
Closed my click miss. Reopening...

PR #24428 (2022-07-16): Adding Steinhart-Hart С coefficient to temperature.cpp

### Description
To resolve  #24243, this request adds:

1. Steinhart-Hart temperature equation "C" coefficient into user temperature sensors parameters inside of Configuration_adv.h
2. Steinhart-Hart temperature equation "C" coefficient initialization to temperature.cpp module.

At this moment, the rest of "C" coefficient handling was already correctly implemented in Temperature::user_thermistor_to_deg_c() method. Only the absence of user defined value for "C" coefficient and corresponding first initialization was a problem.

Nothing of M [... cut]
closed 2022-07-16 by thinkyhead