MarlinFirmware/Marlin #24249: [BUG] SKR Mini E3 V3 stop mid print - only with touch BTT TFT 3.5 V3

rogerrabt, 2022-05-27:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

On Ender-3 Pro, when printing with BTT SKR Mini E3 V3.0 with BTT TFT35 V3.0 attached in touch screen mode, printer stops with heaters on and steppers will not move until powered off.

I can print just fine in Marlin mode, but with the touch screen the print will stop sometime between 5 and 60 minutes.

Versions I have tested with:
* bigtreetech/BIGTREETECH-SKR-mini-E3/tree/master  (crashes in both Touch and Marlin mode)
* MarlinFirmware/Marlin/releases  - version 2.0.9.3  (crashes in Touch, works in Marlin mode)
* MarlinFirmware/Marlin/tree/bugfix-2.0.x  (crashes in Touch, works in Marlin mode)

Hardware variations.  Happens in all 3 cases:
* No BLTouch
* With BLTouch and Z-endstop
* With BLTouch and Z-Homing

Perhaps related to issue #18117?

Attachment includes configs for both 2.0.9.3 and bugfix2.0.x
[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/8783413/Archive.zip)

### Bug Timeline

Started when trying new SKR Mini E3 V3 board

### Expected behavior

Expected print to complete.

### Actual behavior

Printer halts.  Heaters remain on, steppers will not move.  Stop Print does nothing to unlock the situation.

### Steps to Reproduce

Start a print from TFT Touchscreen.

### Version of Marlin Firmware

BIGTREETECH TFT35 V3.0

### Printer model

Creality Ender 3 Pro

### Electronics

BIGTREETECH SKR MINI E3 V3.0

### Add-ons

BLTouch

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

rondlh, 2022-05-28:
Make sure to use the latest TFT firmware (https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware), old versions have some bugs that can cause serial communications issues.

rogerrabt, 2022-05-28:
Thanks rondlh.  I was on TFT35_E3_V3.0.26 and am upgrading to TFT35_V3.0_E3.27 from the release xx.27 branch.
Really hoping this is it!  Thanks for the comment.
(look at all those new features!)

rogerrabt, 2022-05-28:
That did it.  Sorry for the bug.  Closing case

closed 2022-05-28 by rogerrabt