MarlinFirmware/Marlin #24250: [BUG] No fan spin with extruder fan

DoctorDeatch, 2022-05-27:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When printing, the extruder cooling fan does not turn on automatically

### Bug Timeline

it started with the transition to version 2.0

### Expected behavior

the extruder cooling must be turned on when printing or always on

### Actual behavior

the extruder cooling does not turn on automatically, it can only be turned on via the LCD menu or the program

### Version of Marlin Firmware

2.0.9.3

### Printer model

FlyingBear P905H

### Electronics

mks_robin_nano35

### Bed Leveling

ABL 3-point

### Your Slicer

Cura

### Host Software

Repetier Host

### Additional information & file uploads

[https://disk.yandex.ru/d/ze1WUIHS8Uv9sg](url)

thisiskeithb, 2022-05-27:
> #define E0_AUTO_FAN_PIN HEATER_1_PIN
> #define E1_AUTO_FAN_PIN HEATER_1_PIN

You can't set auto fan pins to heater pins.

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2022-05-27 by thisiskeithb

DoctorDeatch, 2022-05-27:
This configuration is already an attempt to solve the problem
Before that, it was set by default to #define E0_AUTO_FAN_PIN -1
but this also doesn't work