MarlinFirmware/Marlin #24254: [BUG] Compiling error: 'thermalManager' was not declared in this scope

kpter, 2022-05-27:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Error while compiling latest `bugfix-2.0.x`.
In Marlin-2.0.9.3 - there are no compile errors.
[configs_and_Diff_last.zip](https://github.com/MarlinFirmware/Marlin/files/8789538/configs_and_Diff_last.zip)

### Version of Marlin Firmware

latest `bugfix-2.0.x` code

### Electronics

BTT SKR 1.3

### Additional information & file uploads

[configs_and_Diff_last.zip](https://github.com/MarlinFirmware/Marlin/files/8789538/configs_and_Diff_last.zip)

```
> Executing task in folder Marlin-bugfix-2.0.9.x: C:\Users\Admin\.platformio\penv\Scripts\platformio.exe run <

Processing LPC1768 (board: nxp_lpc1768; platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.3.zno)
------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nxplpc-arduino-lpc176x/nxp_lpc1768.html
PLATFORM: NXP Arduino LPC176x (0.1.3) > NXP LPC1768
HARDWARE: LPC1768 100MHz, 31.97KB RAM, 464KB Flash
DEBUG: Current (cmsis-dap) On-board (cmsis-dap) External (blackmagic, jlink)
PACKAGES:
 - framework-arduino-lpc176x @ 0.2.9
 - toolchain-gccarmnoneeabi @ 1.100301.220327 (10.3.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ strict
Found 3 compatible libraries
Scanning dependencies...
Dependency Graph
|-- Servo @ 1.0.0
|-- TMCStepper @ 0.7.3
|-- U8glib-HAL @ 0.5.2
Building in release mode
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\HAL.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\HAL_SPI.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\MarlinSerial.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\MinSerial.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\eeprom_flash.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\eeprom_sdcard.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\eeprom_wired.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\fast_pwm.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\include\digipot_mcp4451_I2C_routines.c.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\include\i2c_util.c.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\main.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\tft\tft_spi.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\tft\xpt2046.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\timers.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\u8g\LCD_I2C_routines.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\u8g\LCD_pin_routines.c.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\u8g\u8g_com_HAL_LPC1768_hw_spi.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\u8g\u8g_com_HAL_LPC1768_ssd_hw_i2c.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\u8g\u8g_com_HAL_LPC1768_st7920_hw_spi.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\u8g\u8g_com_HAL_LPC1768_st7920_sw_spi.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\u8g\u8g_com_HAL_LPC1768_sw_spi.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\usb_serial.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\shared\Delay.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\shared\HAL.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\shared\MinSerial.cpp.o
[... 141 lines cut ...]
Compiling .pio\build\LPC1768\src\src\lcd\marlinui.cpp.o
Compiling .pio\build\LPC1768\src\src\lcd\menu\menu.cpp.o
Compiling .pio\build\LPC1768\src\src\lcd\menu\menu_advanced.cpp.o
Compiling .pio\build\LPC1768\src\src\lcd\menu\menu_backlash.cpp.o
Compiling .pio\build\LPC1768\src\src\lcd\menu\menu_configuration.cpp.o
Compiling .pio\build\LPC1768\src\src\lcd\menu\menu_game.cpp.o
Compiling .pio\build\LPC1768\src\src\lcd\menu\menu_main.cpp.o
Compiling .pio\build\LPC1768\src\src\lcd\menu\menu_media.cpp.o
Compiling .pio\build\LPC1768\src\src\lcd\menu\menu_mixer.cpp.o
Compiling .pio\build\LPC1768\src\src\lcd\menu\menu_motion.cpp.o
Compiling .pio\build\LPC1768\src\src\lcd\menu\menu_probe_offset.cpp.o
Compiling .pio\build\LPC1768\src\src\lcd\menu\menu_spindle_laser.cpp.o
Compiling .pio\build\LPC1768\src\src\lcd\menu\menu_tmc.cpp.o
Marlin\src\lcd\menu\menu_motion.cpp: In function 'void _goto_manual_move(const_float_t)':
Marlin\src\lcd\menu\menu_motion.cpp:169:3: error: 'thermalManager' was not declared in this scope
  169 |   thermalManager.set_menu_cold_override(true);
      |   ^~~~~~~~~~~~~~
Compiling .pio\build\LPC1768\src\src\lcd\menu\menu_tune.cpp.o
*** [.pio\build\LPC1768\src\src\lcd\menu\menu_motion.cpp.o] Error 1
==================================================== [FAILED] Took 168.60 seconds =================================

Environment    Status    Duration
-------------  --------  ------------
LPC1768        FAILED    00:02:48.600
================================================ 1 failed, 0 succeeded in 00:02:48.600 ============================```

ellensp, 2022-05-27:
In Marlin/src/lcd/menu/menu_motion.cpp

Replace
```
#if ENABLED(PREVENT_COLD_EXTRUSION)
  #include "../../module/temperature.h"
#endif
```

With
```
#include "../../module/temperature.h"
```
As its is always needed now.

ellensp, 2022-05-27:
Created a PR to fix this.

closed 2022-05-27 by ellensp