MarlinFirmware/Marlin #24257: MINTEMP check not working - SAFETY ISSUE

pillopaolo, 2022-05-28:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

With the heater ON, MINTEMP error is not triggered if I disconnect the thermistor .
Luckily, 20+ seconds later the THERMAL_RUNAWAY error is triggered.

It seems that the issue is due to the wrong calculation of tmin in temperature.cpp, due to a bug in _MAX() macro.

Incriminated line in temperature.cpp: 
const celsius_t tmin = _MAX(HEATER_##NR##_MINTEMP, TERN(TEMP_SENSOR_##NR##_IS_CUSTOM, 0, (int)pgm_read_word(&TEMPTABLE_##NR [TEMP_SENSOR_##NR##_MINTEMP_IND].celsius)));

It should calculate tmin = (5, -15) = 5.
Instead it calculates -15, which is wrong and messes up the MINTEMP detection.

I fixed the issue by calculating the second argument (quite complex) of _MAX() separately, as follows:
const celsius_t tmin_tmp = TERN(TEMP_SENSOR_##NR##_IS_CUSTOM, 0, (int)pgm_read_word(&TEMPTABLE_##NR [TEMP_SENSOR_##NR##_MINTEMP_IND].celsius)); 
const celsius_t tmin = _MAX(HEATER_##NR##_MINTEMP, tmin_tmp); 

Notes: 
a) In my case: HEATER_0_MINTEMP =  5
b) After disconnecting the thermistor I read -15 °C, which is correct, being the lowest temperature in the thermistor table.
c) 2022-05-28 bugfix I see that _MAX() macro has been modified, but even with that fix the above problem is not solved.

### Expected behavior

MINTEMP error should be triggered immediately.

### Version of Marlin Firmware

Marlin-2.0.9.2

thinkyhead, 2022-06-04:
The problem seems to be that `_MAX` has some sign issue, apparently. The -15 was probably appearing as a very large unsigned number. I'll mess around with the `_MIN`/`_MAX` macros / functions and try to get them to behave properly in these cases.

thinkyhead, 2022-06-04:
Which thermistor value were you using? Can you provide your configurations?

thinkyhead, 2022-06-04:
I bet the size of `int` is not 16-bits on some targets, so the `unsigned short` is not being sign-extended after the word is read, giving something like +65521.

The solution is to cast the result of `pgm_read_word` immediately to an `int16_t` instead of the less-well-defined `int`.

closed 2022-06-04 by thinkyhead

pillopaolo, 2022-06-04:
Thanks for the attention and fix. 
However I noted you applied the  int16_t fix to my quick&dirty fix, i.e. the one where I split the calculation in 2 lines.
Why not applying it directly to the original single-line calculation?

commit 9c872b21 (2022-06-04): 🐛 Fix min/max temp evaluation

Fixes #24257
commit 9ab654ad (2022-06-04): 🐛 Fix min/max temp evaluation

Fixes #24257