MarlinFirmware/Marlin #24260: [BUG] Homing Failed on Y axis only

Billy-d-Kidd, 2022-05-29:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Trying to get my X_MAX/MIN_POS and Y_MAX/MIN_POS.
X moves with no problem when issuing the G28 X Y command from Pronterface, but when it's time for Y:
echo:Homing Failed
//action:notification 3D Printer Ready.
//action:notification Homing Failed:
Error:Printer halted. kill() called!
Error:Printer halted. kill() called!

//action:poweroff

I'm not real sure as to where to start looking for this problem, but I have checked hardware, no issues there, I have tried different settings for various settings, and still having this problem.
I am running BTT Octopus 1.1 F446 with TMC2209 drivers using Sensorless Homing.

I have attached my configs for the experts scrutiny, maybe I'm over looking something.
Any help would be great! Thanks!

[Marlin2.0.9.3_configs.zip](https://github.com/MarlinFirmware/Marlin/files/8794388/Marlin2.0.9.3_configs.zip)

### Bug Timeline

Started yesterday (05/28/2022) after downloading the latest bug fix

### Expected behavior

I expect it to home to lower left corner...

### Actual behavior

...killed my printer!

### Version of Marlin Firmware

2.0.9.3

### Printer model

Chiron

### Electronics

BTT Octopus 1.1 F446

Bob-the-Kuhn, 2022-05-30:
Does it resume normal operation when you go back to the old firmware?  If yes, please try to narrow down where the problem started.

Does the Y stepper move at all or is this strictly a sensorless homing issue?

If this is strictly a sensorless homing issue then please enable **TMC_DEBUG** and then issue a **M122 S1** command before doing the G28.  This should give a better indication of what is happening.

Billy-d-Kidd, 2022-05-30:
Y axis movement is good in old firmware, was having problems with Z_SAFE_HOMING that made me get a newer version.
The Y stepper moves. It just seems to be when I'm using G28 or hitting the Home button on the tft.

I have attached the result of M122.
[M122-S0.zip](https://github.com/MarlinFirmware/Marlin/files/8798065/M122-S0.zip)

Bob-the-Kuhn, 2022-05-30:
Sometimes G28 works, sometimes it fails.  

Try adjusting the sensitivity until G28 works reliably.

Billy-d-Kidd, 2022-05-30:
I have sensitivity X=90 and Y=90 right now, but been playing with it from 50 up...any lower and I slam into the frame. Any how, Y is still giving me fits trying to home. I may clone marlin again and start over.

Billy-d-Kidd, 2022-06-01:
> Try adjusting the sensitivity until G28 works reliably.

That seems to have done the trick. Had to set X to 100 and Y to 150, and haven't had a kill signal as of yet.
Thanks Bob-the-Kuhn.

closed 2022-06-01 by thisiskeithb