MarlinFirmware/Marlin #24263: [BUG] enabling #define PSU_CONTROL with TMC2130 drivers causes Marlin not to start

JmanNZ, 2022-05-30:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

If I use #define PSU_CONTROL with TMC2130 drivers Marlin does not start up.
On the serial terminal i get start and then nothing else. (No LCD etc..)
If #define PSU_CONTROL is commented out then Marlin starts as normal.
Mainboard is a MEGA2650   

### Bug Timeline

Not sure only noticed once the TMC2130's were fitted

### Expected behavior

Work as normal

### Actual behavior

Hangs on start

### Steps to Reproduce

1. Ensure TMC2130 drivers in use
2. uncomment #define PSU_CONTROL 
3. Compile and upload
4. Serial terminal will display start and then the board hangs

### Version of Marlin Firmware

02000903

### Printer model

RepRap

### Electronics

Ramps 1.6 Plus and MEGA2650

### Your Slicer

Cura

### Additional information & file uploads

[Configfiles.zip](https://github.com/MarlinFirmware/Marlin/files/8800560/Configfiles.zip)

ellensp, 2022-05-31:
TMC2130 is a stepper driver with SPI communications.
Fairly sure the TMC are not connected to the hardware spi pins  (circuit diagram is not clear)

Enable TMC_USE_SW_SPI In Configuration_adv.h and the ramps no longer locks up. 

Note:Im testing on a ramps 1.4, I cannot test that SPI communications to the tmc's is working

JmanNZ, 2022-05-31:
Thanks for the help
The hardware SPI pins are connected and used.

ellensp, 2022-05-31:
Presuming it is using hardware SPI  

In Marlin/src/MarlinCore.cpp is
```
   #if ENABLED(PSU_CONTROL)
    SETUP_LOG("PSU_CONTROL");
    powerManager.init();
  #endif
```
 powerManager.init(); looks like this

```
void Power::init() {
  psu_on = ENABLED(PSU_DEFAULT_OFF);              // Set opposite state to get full power_off/on
  TERN(PSU_DEFAULT_OFF, power_off(), power_on());
}
```
power_on looks like this

```
void Power::power_on() {
  #if ENABLED(AUTO_POWER_CONTROL)
    const millis_t now = millis();
    lastPowerOn = now + !now;
  #endif

  if (psu_on) return;

  #if EITHER(POWER_OFF_TIMER, POWER_OFF_WAIT_FOR_COOLDOWN)
    cancelAutoPowerOff();
  #endif

  OUT_WRITE(PS_ON_PIN, PSU_ACTIVE_STATE);
  psu_on = true;
  safe_delay(PSU_POWERUP_DELAY);
  restore_stepper_drivers();
  TERN_(HAS_TRINAMIC_CONFIG, safe_delay(PSU_POWERUP_DELAY));

  #ifdef PSU_POWERUP_GCODE
    gcode.process_subcommands_now(F(PSU_POWERUP_GCODE));
  #endif
}

```
This calls  restore_stepper_drivers();

But this code uses SPI which has not been setup yet

Back to  Marlin/src/MarlinCore.cpp we find 

```
  #if HAS_TMC_SPI
    #if DISABLED(TMC_USE_SW_SPI)
      SETUP_RUN(SPI.begin());
    #endif
    SETUP_RUN(tmc_init_cs_pins());
  #endif

```
We need this to be called before 

```
   #if ENABLED(PSU_CONTROL)
    SETUP_LOG("PSU_CONTROL");
    powerManager.init();
  #endif
```

So...
Make this change

```DIFF
--- a/Marlin/src/MarlinCore.cpp
+++ b/Marlin/src/MarlinCore.cpp
@@ -1241,6 +1241,13 @@ void setup() {
     SETUP_RUN(tmc_serial_begin());
   #endif
 
+  #if HAS_TMC_SPI
+    #if DISABLED(TMC_USE_SW_SPI)
+      SETUP_RUN(SPI.begin());
+    #endif
+    SETUP_RUN(tmc_init_cs_pins());
+  #endif
+
   #if ENABLED(PSU_CONTROL)
     SETUP_LOG("PSU_CONTROL");
     powerManager.init();
@@ -1258,13 +1265,6 @@ void setup() {
     SETUP_RUN(disableStepperDrivers());
   #endif
 
-  #if HAS_TMC_SPI
-    #if DISABLED(TMC_USE_SW_SPI)
-      SETUP_RUN(SPI.begin());
-    #endif
-    SETUP_RUN(tmc_init_cs_pins());
-  #endif
-
   SETUP_RUN(hal.init_board());
 
   SETUP_RUN(esp_wifi_init());
```

This doesn't lockup,  but once again I cannot test it, Please try this,

thisiskeithb, 2022-05-31:
Related: https://github.com/MarlinFirmware/Marlin/pull/24160

JmanNZ, 2022-06-01:
Hi
Success confirmed these changes have solved the problem.
Many Thanks

ellensp, 2022-06-01:
I have created a PR to fix this in marlin going forward.
Closing this issue

closed 2022-06-01 by ellensp