MarlinFirmware/Marlin #24267: [BUG] language_it.h  translation use the STR_U, STR_V and STR_W macros that aren't defined

Quasar973, 2022-05-31:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

trying to change language to Italian in configuration.h #define LCD_LANGUAGE it, the compiler reports erros for undefined macros STR_U, STR_V and STR_W that are referred into the language_it.h file but these aren't defined nowere in the code.
trying to change language to Italian in configuration.h #define LCD_LANGUAGE it, the compiler reports erros for undefined macros STR_U, STR_V and STR_W that are referred into the language_it.h file but these aren't defined nowere in the code.

### Bug Timeline

maybe it's new bug

### Steps to Reproduce

1. open file configuration.h 
2. change #define LCD_LANGUAGE en  to #define LCD_LANGUAGE it 
3.Build the project
4. see the errore reported about the missing macros

### Version of Marlin Firmware

2.0.9.3

### Printer model

Creality Ender 3

### Electronics

BTT skr mini E3 v3.0

### Add-ons

none

### Bed Leveling

MBL Manual Bed Leveling

### Your Slicer

Cura

GMagician, 2022-05-31:
Just compiled now, with up to date sources, it compile fine

thisiskeithb, 2022-05-31:
Please please please attach your configs.

ellensp, 2022-05-31:
This has already been fixed in bugfix.

Issue was in Marlin/src/core/language.h
In 2.0.x it is missing the following

```CPP
#if HAS_U_AXIS
  #if AXIS7_NAME == 'U'
    #define STR_U "U"
    #define STR_U_MIN "u_min"
    #define STR_U_MAX "u_max"
  #elif AXIS7_NAME == 'V'
    #define STR_U "V"
    #define STR_U_MIN "v_min"
    #define STR_U_MAX "v_max"
  #elif AXIS7_NAME == 'W'
    #define STR_U "W"
    #define STR_U_MIN "w_min"
    #define STR_U_MAX "w_max"
  #else
    #error "AXIS7_NAME can only be one of 'U', 'V', or 'W'."
  #endif
#else
  #define STR_U   ""
#endif

#if HAS_V_AXIS
  #if AXIS8_NAME == 'V'
    #define STR_V "V"
    #define STR_V_MIN "v_min"
    #define STR_V_MAX "v_max"
  #elif AXIS8_NAME == 'W'
    #define STR_V "W"
    #define STR_V_MIN "w_min"
    #define STR_V_MAX "w_max"
  #else
    #error "AXIS8_NAME can only be one of 'V', or 'W'."
  #endif
#else
  #define STR_V   ""
#endif

#if HAS_W_AXIS
  #if AXIS9_NAME == 'W'
    #define STR_W "W"
    #define STR_W_MIN "w_min"
    #define STR_W_MAX "w_max"
  #else
    #error "AXIS9_NAME can only be 'W'."
  #endif
#else
  #define STR_W   ""
#endif

```

closed 2022-05-31 by ellensp