MarlinFirmware/Marlin #24273: [BUG] Latest bugfix (6/2/2022) causes boot loop

rq3, 2022-06-02:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Todays bug fix (June 2, 2022) compiles fine with the provided configurations. When booting the printer, the display is completely black, and from the sound of the power supply fan, appears to be re-booting about every three seconds.

Changing the display from Classic_ui to Color_ui shows a dim grey display, flashing to bright white as the printer again repetively re-boots about every three seconds.
[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/8824866/Configurations.zip)

### Bug Timeline

Started today, June 2, 2022

### Expected behavior

I expected the printer to boot and function

### Actual behavior

The printer enters a repetitive loop while booting up.

### Steps to Reproduce

1) Compile and load firmware.bin compiled from June 2, 2022 bugfix and example configurations.
2) Turn machine on.
3) Observe repetitive booting attempts.

Sorry, I'm unable to obtain any error logs, as the printer won't boot.

### Version of Marlin Firmware

bugfix June 2, 2022

### Printer model

AnyCubic Predator

### Electronics

trigorilla_pro

### Add-ons

none

### Other things to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.
- [ ] Log output from the host. (`M111 S247` for maximum logging.)
- [ ] Images or videos demonstrating the problem, if it helps to make it clear.
- [ ] A G-Code file that exposes the problem, if not affecting _all_ G-code.

### Additional information & file uploads

Sorry, unable to connect to obtain any log files.

thisiskeithb, 2022-06-02:
Would you be able to test some [`bugfix-2.0.x` commits](https://github.com/MarlinFirmware/Marlin/commits/bugfix-2.0.x) to narrow this down?

The strategy is to find a commit from some point in the "distant" past where the feature works. Then, test a commit from halfway between that date and today…. And then you keep going to the commit half-way in between your "known to work" commit and your "known to be broken" commit until you find the exact day where it broke.

If you started from a point 256 commits in the past, it would take no more than 8 tests to find the exact commit that broke it.

ellensp, 2022-06-02:
This board uses a UART  so you should be able to connect up a simple program to watch the serial port messages even over reboots. 
Eg Pronterface, or even PlatformIO's in built Serial monitor (set monitor_speed = 115200 in platformio.ini to match your Configured baudrate)

thisiskeithb, 2022-06-02:
Replace `Marlin/Marlin/src/HAL/STM32/sdio.cpp` with the one [attached in this zip](https://github.com/MarlinFirmware/Marlin/files/8825424/sdio.cpp.zip) and see if your board boots.

thisiskeithb, 2022-06-02:
This fix may be even easier: https://github.com/MarlinFirmware/Marlin/pull/24275/

```diff
diff --git a/Marlin/src/HAL/STM32/sdio.cpp b/Marlin/src/HAL/STM32/sdio.cpp
index a58589c43ee4..e958d8c3bc48 100644
--- a/Marlin/src/HAL/STM32/sdio.cpp
+++ b/Marlin/src/HAL/STM32/sdio.cpp
@@ -200,7 +200,8 @@ static uint32_t clock_to_divider(uint32_t clk) {
 
 bool SDIO_Init() {
   HAL_StatusTypeDef sd_state = HAL_OK;
-  HAL_SD_DeInit(&hsd);
+  if (hsd.Instance == SDIO)
+    HAL_SD_DeInit(&hsd);
 
   /* HAL SD initialization */
   hsd.Instance = SDIO;
```

jomp16, 2022-06-02:
I can confirm with the above mentioned fix makes Marlin stop boot looping in my Ender 3 Pro.

thisiskeithb, 2022-06-03:
#24275 has been merged.

closed 2022-06-03 by thisiskeithb

dvborzov, 2022-06-11:
fix doesn't work. after the firmware goes into the bootloop, if you pull out the USB flash drive and reboot with power, it starts up.
board  Robin nano v1.3 stm32f103

thisiskeithb, 2022-06-11:
> fix doesn't work. after the firmware goes into the bootloop, if you pull out the USB flash drive and reboot with power, it starts up.

@dvborzov: More work is being done in https://github.com/MarlinFirmware/Marlin/pull/24319