MarlinFirmware/Marlin #24274: BUG  warning: 'void* memset(void*, int, size_t)' clearing an object of non-trivial type 'class SdFile'; use assignment or value-initialization instead [-Wclass-memaccess]

victoroos, 2022-06-02:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When compiling the code in visual studie for the LPC1768 platform I get error:

```
In file included from Marlin\src\sd\../inc/MarlinConfigPre.h:37,
                 from Marlin\src\sd\../inc/MarlinConfig.h:28,
                 from Marlin\src\sd\cardreader.cpp:23:
Marlin\src\sd\cardreader.cpp: In constructor 'CardReader::CardReader()':
Marlin\src\sd\../inc/../core/macros.h:281:49: warning: 'void* memset(void*, int, size_t)' clearing an object of non-trivial type 'class SdFile'; use assignment or value-initialization instead [-Wclass-memaccess]
  281 | #define ZERO(a)             memset(a,0,sizeof(a))
      |                                                 ^
Marlin\src\sd\cardreader.cpp:172:3: note: in expansion of macro 'ZERO'
  172 |   ZERO(workDirParents);
      |   ^~~~
In file included from Marlin\src\sd\cardreader.h:49,
                 from Marlin\src\sd\cardreader.cpp:29:
Marlin\src\sd\SdFile.h:41:7: note: 'class SdFile' declared here
   41 | class SdFile : public SdBaseFile {
      |       ^~~~~~
```

Now the printer isn't able to read the card even though it sees it being attached.

Any tips?
cheers
Vic

### Version of Marlin Firmware

2.0.9.3

### Printer model

Tevo tarantula Pro

### Your Slicer

Cura

### Other things to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.
- [ ] Log output from the host. (`M111 S247` for maximum logging.)
- [ ] Images or videos demonstrating the problem, if it helps to make it clear.
- [ ] A G-Code file that exposes the problem, if not affecting _all_ G-code.

### Additional information & file uploads

[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/8825429/Configuration_adv.zip)

oliver-eifler, 2022-06-02:
You don't use latest bugfix as this 'issue' is already fixed since May 17, 2022
see https://github.com/MarlinFirmware/Marlin/pull/24178
the ZERO macro is now 
#define ZERO(a)             memset((void*)a,0,sizeof(a))

and it's only a warning not an error - nothing which would disturb the function

try real latest bugfix - from today - and see if you still get the warning

thisiskeithb, 2022-06-02:
> You don't use latest bugfix as this 'issue' is already fixed

Configs are way out of date which gives it away as well.

thisiskeithb, 2022-06-02:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/24177

closed 2022-06-02 by thisiskeithb