MarlinFirmware/Marlin #24279: [BUG] M42 does not work first time, but works after that

ryanaukes, 2022-06-03:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

I am using some custom menu items which run M42 commands to switch relays on and off.
When I start up controller and run the M42 command for the first time, it does nothing.
I have to run M42 P112 S255, then M42 P112 S0 (still nothing happens) and after that I can finally use the commands like they are supposed to.

I also tried running the command from pronterface, but it gives the same result.

PS: I am running marlin 2.0.9.2, so not 2.0.9.3. Unfortunately this is the latest firmware version that is published by the makers of the MPCNC for which I am using it.

### Bug Timeline

Today i updated from marlin 2.0.7 to 2.0.9

### Expected behavior

The M42 command working the first time.

### Actual behavior

I have to run M42 P112 S255, then M42 P112 S0 (still nothing happens) and after that I can finally use the commands like they are supposed to.

### Steps to Reproduce

1. Enable DIRECT_PIN_CONTROL in the Configuration_adv.h file
2. Flash Marlin to your mainboard
3. Try to run M42 after you started your mainboard

### Version of Marlin Firmware

2.0.9.2

### Printer model

MPCNC

### Electronics

SKR Pro 1.2

### Add-ons

Relays connected to pin 46 and 112

### Your Slicer

Other (explain below)

### Host Software

SD Card (headless)

### Other things to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.
- [ ] Log output from the host. (`M111 S247` for maximum logging.)
- [ ] Images or videos demonstrating the problem, if it helps to make it clear.
- [ ] A G-Code file that exposes the problem, if not affecting _all_ G-code.

### Additional information & file uploads

Since this is about a CNC machine i dont u a slicer, but I use Lightburn or ESTLCam

[Configuration files.zip](https://github.com/MarlinFirmware/Marlin/files/8831763/Configuration.files.zip)

thisiskeithb, 2022-06-03:
> PS: I am running marlin 2.0.9.2, so not 2.0.9.3. Unfortunately this is the latest firmware version that is published by the makers of the MPCNC for which I am using it.

We can’t support forks of Marlin here, so you’ll need to file a bug with them.

ryanaukes, 2022-06-03:
> > PS: I am running marlin 2.0.9.2, so not 2.0.9.3. Unfortunately this is the latest firmware version that is published by the makers of the MPCNC for which I am using it.
> 
> 
> 
> We can’t support forks of Marlin here, so you’ll need to file a bug with them.

Thanks for your reply.
I don't think it can be called a fork though, since they just provide the right configuration, they dint modify the firmware itself.

Bob-the-Kuhn, 2022-06-03:
I just tried it on a SKR PRO 1.1 and M42 worked the first time.  This was using the latest bugfix 2.0.x

You could add your initialization commands to the **STARTUP_COMMANDS** macro (in configuration_adv.h).

ryanaukes, 2022-06-15:
> I just tried it on a SKR PRO 1.1 and M42 worked the first time.  This was using the latest bugfix 2.0.x
> 
> 
> 
> You could add your initialization commands to the **STARTUP_COMMANDS** macro (in configuration_adv.h).

Thank you for your reply! Which pins did you use? Maybe it has something to do with the pins I am trying to use.