MarlinFirmware/Marlin #24281: [BUG] V86 echo message triggered during print

Mrlonch, 2022-06-03:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

Print pauses randomly, usually in big prints over 5hrs. Triggering the V86 echo message similar to when EEPROM settings are stored in a slot. Print cannot be resumed as printer loses height reference.

### Bug Timeline

It first started after upgrading to 2.0.9.3

### Expected behavior

I expect the print to finish as any other printing parameter seems ok.

### Actual behavior

It randomly pauses print at any time, mainly experienced with bigger prints or long lasting ones. It shows stored settings retrieved message in the LCD with V86 echo code.

I had prints stoping in the first layer and some other after 8hrs printing.

### Steps to Reproduce

Run a print normally

### Version of Marlin Firmware

2.0.9.3

### Printer model

Artillery Genius

### Electronics

MKS Gen 1

### Add-ons

Bl touch from antlabs

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Other (explain below)

### Other things to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.
- [ ] Log output from the host. (`M111 S247` for maximum logging.)
- [ ] Images or videos demonstrating the problem, if it helps to make it clear.
- [ ] A G-Code file that exposes the problem, if not affecting _all_ G-code.

### Additional information & file uploads

Slicer I am current using is Superslicer in its latest version

thisiskeithb, 2022-06-03:
> Other things to include
> A ZIP file containing your Configuration.h and Configuration_adv.h.

You did not do this. Please attach your configs.

Mrlonch, 2022-06-04:
Attached! Sorry!

Yes I did update to latest TFT firmware as well, the one from Digant. The filament runout error went away, but this one still persist and its driving me crazy, last night print failed after two hours. Printer pauses, hotend and bed go to 0 and the nozzle just lifts a couple of centimeters and thats it, cannot recover the print despite it seems to be possible... I continuously get the message V86... any hint?

digant73, 2022-06-08:
@Mrlonch Did you try to use my Marlin 2.0.9.2 already configured for SW1? Did you have the same issue with that?

Mrlonch, 2022-06-08:
> @Mrlonch Did you try to use my Marlin 2.0.9.2 already configured for SW1? Did you have the same issue with that?

Yes, but with it I get thermal runaway error, that is why I went for 2.0.9.3, but I may try one more time to go back again as I managed to set the correct K values for the PID

digant73, 2022-06-08:
built-in PID values were for SW1. So yes, for a Genius after Marlin fw installation you always need to perform EEPROM reset, PID, BL. I had no issue at all.

Mrlonch, 2022-06-08:
I’ll try again and update later today! 

Thanks for your time!

Mrlonch, 2022-06-09:
Went back to 2.0.9.2 with the adequate PID K indexes and completed a print that was failing so far!! Still do not know what’s the issue on 2.0.9.3...

thisiskeithb, 2022-06-23:
Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

Mrlonch, 2022-06-30:
> Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

I installed 2.1 succesfully, it seems the problem has gone, although I consider I still have some bed issues to sort... yesterday got a print stop with V86 echo message and the bed sensor was reading -1. I have ordered heater mat replacement, I will update after installing it and confirm no further print stops happen.... lots of wasted material already

but thanks to the community! thanks for your time and support!