MarlinFirmware/Marlin #24292: [BUG] Display does not work on K8800 board because contrast range in pins_K8800.h appears to be wrong.

sgparry, 2022-06-05:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When using firmware built with:
```#define MOTHERBOARD BOARD_K8800```
the display does not work. This can be tracked down to the following lines in pins_K8800:
```  #define LCD_CONTRAST_MIN                     0
  #define LCD_CONTRAST_MAX                   100
  #define LCD_CONTRAST_INIT                   30```
if these are removed, the display works properly with the defaults.

There is some disagreement as to which is the closest display match for the K8800; Velleman's own customisation of Marlin 1.1.4, used ```ELB_FULL_GRAPHICAL_CONTROLLER```, but this requires a customised version of u8g. Other developers, e.g. Susisstrollch, have specified the  ```AZSMZ_12864``` which seems to be the closest match. Either way, the contrast range specified in pins_K8800 does not work and probably does not belong there.

### Bug Timeline

Since 22-Sep-2020 225487f9db7d1a7aaf6886a283c7758a9ae2181e

### Expected behavior

When built with ```#define AZSMZ_12864```, display should be visible.

### Actual behavior

When built with ```#define AZSMZ_12864```, display is only just visible at max contrast. At default contrast the display appears blank.

### Steps to Reproduce

Setup PlatformIO / Visual Studio Code for command line build.
Clone source code.
Replace Configuration.h, Configuration_adv.h with those below.
Build
Flash K8800 Printer with newly built firmware.
Restart printer.
Observe blank display.

### Version of Marlin Firmware

Recv: FIRMWARE_NAME:Marlin bugfix-2.1.x (Jun  5 2022 22:07:09) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Vertex Delta EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff

### Printer model

Velleman Vertex Delta K8800

### Electronics

Stock

### Add-ons

None

### Bed Leveling

UBL Bilinear mesh

### Host Software

OctoPrint

### Other things to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.
- [ ] Log output from the host. (`M111 S247` for maximum logging.)
- [ ] Images or videos demonstrating the problem, if it helps to make it clear.
- [ ] A G-Code file that exposes the problem, if not affecting _all_ G-code.

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/8840803/Configuration.zip)

thinkyhead, 2022-10-14:
Just catching up on missed issues… Currently there is no mention of contrast in the K8800 file.

Please test the [`bugfix-2.1.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

thisiskeithb, 2022-10-14:
> Just catching up on missed issues… Currently there is no mention of contrast in the K8800 file.

This was fixed in the referenced issue https://github.com/MarlinFirmware/Marlin/pull/24402

closed 2022-10-14 by thisiskeithb