MarlinFirmware/Marlin #24295: [BUG] Build failed with MESH_BED_LEVELING enabled

Sergey1560, 2022-06-06:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

If MESH_BED_LEVELING is enabled, the build failed with error:
error: 'class mesh_bed_leveling' has no member named 'get_z';

Marlin/src/core/utility.cpp line 159 and line 163.

The "mesh_bed_leveling" class does not have a method "get_z" in the Marlin/src/feature/bedlevel/mbl/mesh_bed_leveling.h file. Perhaps this should be the method "get_z_correction()" or "get_z_offset()" ?

### Bug Timeline

This is a new bug

### Expected behavior

Build without errors.

### Actual behavior

Build failed.

### Steps to Reproduce

1. Enable #define MESH_BED_LEVELING in Marlin/Configuration.h
2. Build firmware

### Version of Marlin Firmware

2.1.0

### Bed Leveling

MBL Manual Bed Leveling

### Other things to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.
- [ ] Log output from the host. (`M111 S247` for maximum logging.)
- [ ] Images or videos demonstrating the problem, if it helps to make it clear.
- [ ] A G-Code file that exposes the problem, if not affecting _all_ G-code.

### Additional information & file uploads

[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/8844055/configuration.zip)

tombrazier, 2022-06-06:
I know what is causing this. Will submit a PR. I suspect 2.0.9.4 and 2.1 will both need it.

thisiskeithb, 2022-06-06:
Closing since a PR was opened: https://github.com/MarlinFirmware/Marlin/pull/24297

closed 2022-06-06 by thisiskeithb