MarlinFirmware/Marlin #24296: [BUG] (Coordinates suddenly shifted in 2.0.9.4, 2.1 and bugfix-2.1.x)

unclej84, 2022-06-06:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

A previously running gcode suddently turned a movement in the wrong directions - position reportings show strange jump in position values. So instead of moving down to a wanted z-position the gantry on the ender style printer moves up into the frame and collides. In 2.0.9.3 the gcode works as intended

### Bug Timeline

working: bugfix-2.0.x (18.02.2022 20:05), not working releases 04.06.2022

### Expected behavior

No change in coordinates without any action

### Actual behavior

Moved up instead of down

### Steps to Reproduce

Simply running attached gcode via terminal or sd-card - same result in both cases - the gantry moves up instead of down

### Version of Marlin Firmware

2.0.9.3

### Printer model

Creality Ender-3 V2

### Electronics

Octopus V1.1

### Add-ons

dual lead screws

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

### Other things to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.
- [ ] Log output from the host. (`M111 S247` for maximum logging.)
- [ ] Images or videos demonstrating the problem, if it helps to make it clear.
- [X] A G-Code file that exposes the problem, if not affecting _all_ G-code.

### Additional information & file uploads

[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/8845481/configuration.zip)
[Test.gcode.zip](https://github.com/MarlinFirmware/Marlin/files/8845496/Test.gcode.zip)
[Console_Octoprint.txt.zip](https://github.com/MarlinFirmware/Marlin/files/8845497/Console_Octoprint.txt.zip)

SlevinSkrreeek, 2022-07-12:
Same here with voxelab aquila on bugfix-2.1.x and UBL enabled. undef SEGMENT_LEVELED_MOVES fix the problem.

unclej84, 2022-07-21:
Did you experience any negative influence upon setting the undef?

unclej84, 2022-09-20:
@rainberd, @SlevinSkrreeek have you tried 2.1.1? Is it fixed there?

rainberd, 2022-10-19:
I can't reproduce on bugfix-2.1.x ([e49c3dc](https://github.com/MarlinFirmware/Marlin/commit/e49c3dc0889f1a6b597701ceb69624bdf4365445))

SlevinSkrreeek, 2022-10-19:
> @rainberd, @SlevinSkrreeek have you tried 2.1.1? Is it fixed there?

Sorry, didn't find time to rebuild the firmware recently

unclej84, 2022-10-23:
The current bugfix-2.1.x ([aaf34fa](https://github.com/MarlinFirmware/Marlin/commit/aaf34fa96bf3da9cf7c95a41ccb7e5bc441dbd20)) seems to work for me too. Hope the next released version will stay the same

closed 2022-10-23 by thisiskeithb