MarlinFirmware/Marlin #24300: [FR] Multi-home Z

jmding8, 2022-06-07:
### Is your feature request related to a problem? Please describe.

I have a Kingroon KP3s. For some reason, it's z-home position is unreliable. Probably 1 in 6 prints it will z-home roughly 0.4mm too low, resulting in essentially no first layer at all due to the nozzle being pressed fully in contact with the bed. The rest of the time, it homes to a fairly consistent location and produces a good first layer.

I've inverted the Z endstop from its stock NO configuration to NC, and this has not improved things.

It's not clear to me why this is happening, but it seems plausible that it is some sort of EMI.

In any case, while looking for a solution I've seen other reports of similar phenomena from other users.

### Describe the feature you want

Multi-home z using a similar approach to what is done for multi-probing and automated bed leveling:

1. Home Z multiple times
2. Reject extreme outliers
3. Average the other results.

ellensp, 2022-06-07:
/**
 * Endstop Noise Threshold
 *
 * Enable if your probe or endstops falsely trigger due to noise.
 *
 * - Higher values may affect repeatability or accuracy of some bed probes.
 * - To fix noise install a 100nF ceramic capacitor in parallel with the switch.
 * - This feature is not required for common micro-switches mounted on PCBs
 *   based on the Makerbot design, which already have the 100nF capacitor.
 *
 * :[2,3,4,5,6,7]
 */
//#define ENDSTOP_NOISE_THRESHOLD 2

jmding8, 2022-06-07:
Thanks for this, I'll experiment with this setting and report back.

rondlh, 2022-06-08:
You can try lowering the probing speed:
Z_PROBE_FEEDRATE_FAST
Z_PROBE_FEEDRATE_SLOW

and/or use multi-probing:

/**
 * Multiple Probing
 *
 * You may get improved results by probing 2 or more times.
 * With EXTRA_PROBING the more atypical reading(s) will be disregarded.
 *
 * A total of 2 does fast/slow probes with a weighted average.
 * A total of 3 or more adds more slow probes, taking the average.
 */
//#define MULTIPLE_PROBING 2
//#define EXTRA_PROBING    1

thisiskeithb, 2022-06-08:
> Z_PROBE_FEEDRATE_FAST
> Z_PROBE_FEEDRATE_SLOW
> and/or use multi-probing

@rondlh: That only works when using a probe. OP is homing Z using a Z endstop.

jmding8, 2022-06-09:
I've tried in that in the past, even with 1/16x z-homing speed, the occasional miss-homing still happened

jmding8, 2022-07-01:
Unfortunately, ` #define ENDSTOP_NOISE_THRESHOLD 2` did not improve homing consistency. I even replaced the endstop again with one that is known-good. I wonder if it's possible that I have a hardware issue with my main board? Still, there may be other users experiencing similar issues, where multi-homing would be of value

jmding8, 2022-07-23:
I replaced my mainboard and my endstop switch and am still experiencing inconsistent z homing position, both with and without `ENDSTOP_NOISE_THRESHOLD`. At this point, I suspect that it is actually something with Marlin given that I've basically replaced everything else that could be the problem. Lowering homing speed also has not helped. @ellensp any suggestions?

descipher, 2022-07-24:
This could be a physical problem, 0.4mm is quite large to be a probe noise issue. Set the z probing location as close to the z post and re-test it. If it minimizes the issue then it is related to the arm dropping down due to slop on the z-post rollers.

jmding8, 2022-07-24:
I'm not sure what you described applies. I'm not using a probe, this is traditional endstop based homing on a Kingroon kp3s. The endstop is essentially coaxial with the z rollers, there shouldn't really be any arm to drop.

descipher, 2022-07-24:
Right, if its endstop switch noise there would be homing deviations on x or y as well. Have you verified belt tensions are good. Also with two motors to lift on z I would make sure the z stepper is driving at the max safe current. Could also be some z extra belt backlash with the heavy two motor load.

thisiskeithb, 2022-07-24:
> Also with two motors to lift on z

I’m pretty sure this is a single Z printer: https://youtu.be/XR0hI4pVvlY

descipher, 2022-07-24:
Yes it is, I am referring to the mass of the z and x axis motors for the one z motor to lift.

thisiskeithb, 2022-07-24:
> Describe the feature you want
> Multi-home z using a similar approach to what is done for multi-probing and automated bed leveling

With `MULTIPLE_PROBING`, we still expect the probe to trigger and average the results (and with `EXTRA_PROBING`, it will toss atypical readings).

Your Z endstop seems to randomly fail & not trigger at all, so I'm not sure how that can be dealt with other than the current behavior of halting (assuming you have `VALIDATE_HOMING_ENDSTOPS` enabled).

jmding8, 2022-07-24:
> Right, if its endstop switch noise there would be homing deviations on x or y as well.

Have not verified this because it doesn't typically come up in my day-to-day printing. This would only come up for me in the case of power-outage recovery. I could test it, but it doesn't really seem particularly productive.

> Have you verified belt tensions are good.

Yes, they are good. Although in this case, since the Z is lead screw driven and we home with X and Y static and in a consistent position, I don't see the relevance? 

> Also with two motors to lift on z I would make sure the z stepper is driving at the max safe current.

I haven't verified this, but I have my doubts that this would be the issue, between the mechanical advantage of the lead screw design and the fact that I'm not experiencing any skipped-step issues on Z. 

> Could also be some z extra belt backlash with the heavy two motor load.

Lead screw z, so it's not belt related. Could be backlash, but that's unlikely because backlash would also be consistent. Also, essentially all Ender 3 style printers carry more mass on the single z stepper than the KP3s does, the bowden extruder motor is still mounted to the X gantry

jmding8, 2022-07-24:
> Your Z endstop seems to randomly fail & not trigger at all, so I'm not sure how that can be dealt with other than the current behavior of halting (assuming you have VALIDATE_HOMING_ENDSTOPS enabled).

Halting and alerting the user would be a good option IMO.

Separately, even in the case where the endstop randomly fails and triggers 0.5mm too late every once in a while, some sort of averaging algorithm would be better than not averaging. Another option could be to continually re-probe until the running average converges: probe 3 times, and keep probing until each additional probe does not change the running average by more than 0.01mm or some other low threshold.

descipher, 2022-07-24:
Ok, lets assume we have noise path. Have you verified the ground return path to the mains supply? What model and make is the PSU? Share images of the supply junctions to ground?
Please note that the probability of a code issue on this issue is nearly 0.
Also the switch type in the observable images would require on board noise clamping. If the original design incorporated it then a replacement of the MB with out it would require external noise clamping.

jmding8, 2022-07-25:
@descipher thanks for the suggestions. The majority of them went over my head, I'll do my best to answer.

This is the power supply, ZL-360-24. Seems to be relatively generic.

[image]

It is connected to the main board via a barrel jack:

[image]

This is the main board, a Kingroon specific version of the MKS Robin Nano. Both the original and replacement boards are this one:

[image]

I am not sure how to verify the ground path to the mains supply. Probe for continuity between the endstop and any ground point on the main board?

I'm not sure how to verify if this board has on-board noise clamping. Any suggestions?

thisiskeithb, 2022-07-25:
> Halting and alerting the user would be a good option IMO.

That is the current behavior.

If your endstops do not trigger within the expected steps / travel limits & you have `VALIDATE_HOMING_ENDSTOPS` enabled, Marlin will halt.

Since this is a hardware issue and not something that can be corrected in firmware, please use these use these resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

closed 2022-07-25 by thisiskeithb

descipher, 2022-07-25:
PSU is Ok. Board you have does have noise clamping. There appears to be no ground path to mains in the design that I can see.
You can add a ground wire from the gantry base assembly by securing it to a screw contacting a cleaned bare metal location of it that is near the controller board. The wire should enter the PSU AC junction box and connect to the AC ground terminal which is the middle terminal of the C13 plug and it should already be attached to the ground terminal of the PSU. Keep in mind that sometimes the AC source may not be properly grounded, rare but possible.

[image]
AC C13 Ground Terminal
[image]
PSU Ground Terminal

Other observations, The switch wires are to close the X stepper motor and will induce a signal.
[image]

jmding8, 2022-07-27:
I have updated my starting gcode to
Home xy
Disable Motors 
Home z
ReHome xy
Turn on heaters 

This seems to be helping. If I don't update again, then it likely means the issue was fixed