MarlinFirmware/Marlin #24301: [BUG] LCD freezes/very sluggish after upgrade if SD card is left in the mainboard

ucirello, 2022-06-07:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Immediately upon upgrade to 2.0.9.4, then 2.1, and bugfix-2.1 (371ac49fc137fbae6bf58491e274d1099ddafa34), the LCD freezes, or at least gets very sluggish. 

It worked fine with 2.0.9.3.

### Bug Timeline

Since 2.0.9.4 - did not bisect in search for the first offending commit

### Expected behavior

I expected the LCD to render quickly and be able to navigate through all menus. Turning the knob clockwise or counter-clockwise would immediately change feed rate.

### Actual behavior

Depending on the boot sequence it will either freeze completely, or be very sluggish.

The info screen renders in more than 15s per half. 

The wheel button seems to not be able to open the main menu.

Turning the knob clockwise or counter-clockwise would immediately change feed rate affects the feed rate, but it takes a long time for the number to update.

### Steps to Reproduce

1. Compile bugfix-2.1 (371ac49fc137fbae6bf58491e274d1099ddafa34) with the given configuration files.
2. Use a LPC1769 board (tested with MKS sgenl v2) with a RepRap Full Graphical Display and flash the firmware.
3. LCD freezes.

### Version of Marlin Firmware

FIRMWARE_NAME:Marlin bugfix-2.1.x (Jun  6 2022 18:18:47) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0

### Printer model

Modded Ender 3

### Electronics

MKS SGENL v2, Discount Full Graphic Smart Controller

### Add-ons

EZNEO, EZABL, UART based serial connection

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Other things to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.
- [ ] Log output from the host. (`M111 S247` for maximum logging.)
- [X] Images or videos demonstrating the problem, if it helps to make it clear.
- [ ] A G-Code file that exposes the problem, if not affecting _all_ G-code.

### Additional information & file uploads

[bugfix-2.1-config.zip](https://github.com/MarlinFirmware/Marlin/files/8849270/bugfix-2.1-config.zip)

[image]
[image]

ellensp, 2022-06-07:
I'm looking at a similar issue on a 1284p  https://github.com/MarlinFirmware/Marlin/issues/24298
Can you test disabling sdsupport as a test ?
This stop the issue on the 1284p, not a solution, but a big clue

ucirello, 2022-06-07:
@ellensp @thisiskeithb I bisected and the culprit seems to be:

```
15b6159f6af740f17db2dc8815697759c714cc12 is the first bad commit
commit 15b6159f6af740f17db2dc8815697759c714cc12
Author: Robby Candra <robbycandra.mail@gmail.com>
Date:   Mon May 23 09:13:17 2022 +0700

    🩹 Adjust manage_media for slow/late media init (USB FD) (#24015)

 Marlin/src/sd/cardreader.cpp | 38 +++++++++++++++++++++++++-------------
```

ellensp, 2022-06-07:
@ucirello see ssue https://github.com/MarlinFirmware/Marlin/issues/24298

ellensp, 2022-06-07:
closing this as a duplicate of https://github.com/MarlinFirmware/Marlin/issues/24298

closed 2022-06-07 by ellensp