MarlinFirmware/Marlin #24306: [BUG] LCD freezes/very sluggish after upgrade if SD card is left in the mainboard (continues #24301)

ucirello, 2022-06-07:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

With 3e9fb34892e85bc4069acf5baddbf12d6cd47789 still, the LCD freezes, or at least gets very sluggish.

It worked fine with 2.0.9.3.

### Bug Timeline

Since 15b6159f6af740f17db2dc8815697759c714cc12 it is broken. 3e9fb34892e85bc4069acf5baddbf12d6cd47789 did not fix.

### Expected behavior

I expected the LCD to render quickly and be able to navigate through all menus. Turning the knob clockwise or counter-clockwise would immediately change feed rate.

### Actual behavior

Depending on the boot sequence it will either freeze completely, or be very sluggish.

The info screen renders in more than 15s per half.

The wheel button seems to not be able to open the main menu.

Turning the knob clockwise or counter-clockwise would immediately change feed rate affects the feed rate, but it takes a long time for the number to update.

### Steps to Reproduce

1. Compile bugfix-2.1 (3e9fb34892e85bc4069acf5baddbf12d6cd47789) with the given configuration files.
2. Use a LPC1769 board (tested with MKS sgenl v2) with a RepRap Full Graphical Display and flash the firmware.
3. LCD freezes.

### Version of Marlin Firmware

bugfix-2.1 (3e9fb34892e85bc4069acf5baddbf12d6cd47789)

### Printer model

Modded Ender 3

### Electronics

MKS SGENL v2, Discount Full Graphic Smart Controller

### Add-ons

EZNEO, EZABL, UART based serial connection

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Other things to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.
- [ ] Log output from the host. (`M111 S247` for maximum logging.)
- [X] Images or videos demonstrating the problem, if it helps to make it clear.
- [ ] A G-Code file that exposes the problem, if not affecting _all_ G-code.

### Additional information & file uploads

[image] (photo taken this morning)

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/8853725/configs.zip)

[image]

ucirello, 2022-06-07:
Also related to https://github.com/MarlinFirmware/Marlin/issues/24298

ne0nex, 2022-06-07:
Interestingly, this is the opposite that I'm having happen on my SKR1.3 board in https://github.com/MarlinFirmware/Marlin/issues/24293

If an SD card is *not* inserted (and power loss recovery is set to on) at boot, the machine gets into this same state.

ucirello, 2022-06-07:
I think part of the problem is now that there are two SD slots. One in the mainboard, and another in the display. This behavior only happens when the SD is in the mainboard's slot. I didn't try in the LCD slots to see what happens.

ne0nex, 2022-06-07:
Interesting. My display doesn't have an SD slot. I only have the on-board.

willngton, 2022-06-08:
Exactly same issue with MKS Robin (KP3S)

Would work fine if boot without SD Card in.

https://user-images.githubusercontent.com/5961905/172507243-4a5d037b-df7f-4c2c-9c86-e2f05130c8b1.mp4

ellensp, 2022-06-08:
I cannot replicate this on a btt skr 1.3 with ucirello configs (with motherboard updated)
or on a btt skr 1.4 turbo with ucirello configs (with motherboard updated)
Using current Marlin bugfix 2.1.x

Tried all 4 physical combinations of sdcard and powered on.

```
onboard    | in LCD  | status
--------------------------------
empty      | empty   | no issues
empty      | sdcard  | no issues
sdcard     | empty   | no issues
sdcard     | sdcard  | no issues
```

Doubledeucedavis, 2022-06-08:
I am not on the Marlin Bugfix 2.1.x. I am on Release 2.1

ellensp, 2022-06-08:
@Doubledeucedavis Release is a snapshot in time. It generally does not get updated after it is made.
It does not have patch https://github.com/MarlinFirmware/Marlin/pull/24302 installed

You need to use either Marlin 2.1.x https://github.com/MarlinFirmware/Marlin/archive/2.1.x.zip
or  bugfix 2.1.x https://github.com/MarlinFirmware/Configurations/archive/refs/heads/bugfix-2.1.x.zip
or apply the patch yourself to the release code

ellensp, 2022-06-08:
Played some more...

On a skr 1.4 with a RepRap Full Graphical Display,  If I put a empty Micro SDCARD TO SDCARD adapter into the lcd sdcard slot this will make it fail.
Ie the card detect IO line says there is SD card inserted, but it is unreadable (in this case because it is not present) 

The 128x64 screen does (eventually) say "Media Init Fail" and the serial log shows "echo: No SD card" over and over, and the display is all slow.

@willngton I do notice that your display also says "Media Init Fail"

@ucirello Can you give you display enough time to fully render and check if you also have a "Media Init Fail" message?

So perhaps this issue is that Marlin is not failing gracefully when the sdcard will not mount but is detected as present.

mametaro999, 2022-06-08:
I have the same issue on Kingroon KP3 (original MKS Robin mini board).  

I compared source tree between build 2.0.9.3 and 2.1.0.0 and found \Marlin-bugfix-2.1.x\Marlin\src\HAL\STM32\sdio.cpp was changed to use CMSIS library.
Then I tried to roll back this file to 2.0.9.3 in 2.1.0.0 source and confirmed this issue was gone with old source code to use old STM32 library.
So I think sdio.cpp may be incomplete for STM32F1 board support, if moving it to use CMSIS library.
Thanks.

ucirello, 2022-06-08:
@ellensp when I use `cardreader.cpp` from 04d0d9431b, it seems to work.

ellensp, 2022-06-10:
The PR should fix this case of slow updating displays. https://github.com/MarlinFirmware/Marlin/pull/24319
It wont address the underlying issue of not being able to mount the sdcard. It just handles that condition is a much nicer way.

Fannattatic, 2022-06-20:
I had a similar issue on my SKR mini V2 where the screen would take ages to load but I solved it by using the maple build environment (STMF103RE_btt_USB_maple (512K)) instead of the standard build environment (I used STM32F103RE_btt_USB (512K)). I have no idea why this makes a difference. 

Also yes maple is now deprecated but it works for me.

dburr, 2022-06-21:
MKS Robin nano v1.2 with MKS TFT35
Seeing a problem similar to @willngton - if I boot up with an SD card inserted, the mainboard bootloops. However if I boot without an SD card inserted, the machine comes up fine, albeit with a somewhat sluggish LCD. If I then plug in an SD card and run a print, the print runs for a while but then it fails (either reboots or just plain locks up.) Have tried both 2.1 as well as bugfix-2.1.x as of 20220618.

thisiskeithb, 2022-06-21:
https://github.com/MarlinFirmware/Marlin/pull/24319 has been merged (with some changes from when it was initially submitted) so please try the latest [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/refs/heads/bugfix-2.1.x.zip) and report back.

thisiskeithb, 2022-06-23:
To answer my own suggestion above:

With the following basic config running the current `bugfix-2.1.x` (1a79537), I get "Media Init Fail", but no LCD sluggishness.

```cpp
#define MOTHERBOARD BOARD_MKS_SGEN_L_V2
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
#define SDCARD_CONNECTION LCD
```

Changing `SDCARD_CONNECTION LCD` to `SDCARD_CONNECTION ONBOARD` works, but obviously the LCD-based SD slot isn't enabled/in use.

ucirello, 2022-06-23:
> Changing SDCARD_CONNECTION LCD to SDCARD_CONNECTION ONBOARD works, but obviously the LCD-based SD slot isn't enabled/in use.

That tracks with my own experience here.

thisiskeithb, 2022-06-23:
Some more testing/swapping in another LCD:

`#define REPRAP_DISCOUNT_SMART_CONTROLLER` (2004 character LCD) works fine with `#define SDCARD_CONNECTION LCD`.

thinkyhead, 2022-07-05:
Please test the [`bugfix-2.1.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further — and possibly revert the media handling code to an earlier state.

ucirello, 2022-07-06:
It works with the configuration below at 132c37a6511c89306a60c7663265aef257b007f5:
Configuration.h
```cpp
#define MOTHERBOARD BOARD_MKS_SGEN_L_V2
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
#define SDSUPPORT
````

Configuration_adv.h
```cpp
#define SDCARD_CONNECTION LCD
```

platformio.ini
```ini
[platformio]
#...
default_envs = LPC1769
```

two small notes:

1 - It does seem that the beep on start is taking longer when the card is not present.
2 - when I hit `Change Media` it never indicates whether is unmounted or not. When I pull the SD card off the LCD 12864 display, it hangs for a couple of seconds, and only then it shows in the menu "No Media"

thisiskeithb, 2022-07-06:
This may be a solution: https://github.com/MarlinFirmware/Marlin/issues/22898#issuecomment-1175972115

closed 2024-02-21 by ucirello