MarlinFirmware/Marlin #24308: [BUG] Hotend fan spinning very slow at 100% / fan or fw problem?

lipewww, 2022-06-08:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I'm having a problem with the hotend fan since I updated to marlin 2.0.9.3, but I don't know if the problem is with the fan or the fw. When I turn my fan on 100%, it spins slowly (maybe 40% of the speed). Sometimes it even makes some noise. And when reducing to 50% for example, the cooler can't even spin so slowly anymore.

I decided to invert the connection between the hotend fan and the extruder fan on the motherboard, and when turning on the machine, the fan turned on very slowly in the same way, and the extruder fan at 100% working perfectly.

Almost certain that the problem is in the fan?

### Bug Timeline

Since 2.0.9.3

### Version of Marlin Firmware

2.1 bugfix

### Printer model

Anet A6

### Bed Leveling

No Bed Leveling

### Your Slicer

Cura

### Host Software

Cura

### Other things to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.
- [ ] Log output from the host. (`M111 S247` for maximum logging.)
- [ ] Images or videos demonstrating the problem, if it helps to make it clear.
- [ ] A G-Code file that exposes the problem, if not affecting _all_ G-code.

### Additional information & file uploads

[config marlin 2.1 bugfix.zip](https://github.com/MarlinFirmware/Marlin/files/8860763/config.marlin.2.1.bugfix.zip)

GMagician, 2022-06-08:
You may connect your fan to 12V, 24V or 5V directly (see fan voltage) and see if it works.

lipewww, 2022-06-08:
> You may connect your fan to 12V, 24V or 5V directly (see fan voltage) and see if it works.

It is 12v, but as I said, I connected the extruder fan and it turned on at high speed normally, the hotend fan, which was supposed to stay on all the time at 100%, was still slow. Anyway, I'm already waiting for a new fan to arrive to test.

lipewww, 2022-06-11:
It was a problem fan

closed 2022-06-11 by lipewww