MarlinFirmware/Marlin #24321: [BUG] M569 S0/1 stealthChop reversed on X

fuzebox1, 2022-06-10:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Stealthchop wouldn't turn on for me on X, while on all other axis it would.
So i tried M569, it reported off on X, i did M569 S1 X to turn it on. Still off.
Tried S0, turned it on.
Tried on other axis, seems reversed.

My terminal output:
`_All steppers on stealth._
Send: M569
Recv: X driver mode:	stealthChop
Recv: Y driver mode:	stealthChop
Recv: Z driver mode:	stealthChop
Recv: Z2 driver mode: stealthChop
Recv: E driver mode:	stealthChop
Recv: ok
[...]
_S0 on Y turns Y to spread._
Send: M569 S0 Y
Recv: ok
[...]
Send: M569
Recv: X driver mode:	stealthChop
Recv: Y driver mode:	spreadCycle
Recv: ok
[...]
_S1 on Y turns stealth back on._
Send: M569 S1 Y
Recv: ok
[...]
Send: M569
Recv: X driver mode:	stealthChop
Recv: Y driver mode:	stealthChop
_S1 on X turns X to spread. Reversed on X?!_
Send: M569 S1 X
Recv: ok
[...]
Send: M569
Recv: X driver mode:	spreadCycle
Recv: Y driver mode:	stealthChop
_S0 turns stealth on again. Again, reversed?!_
Send: M569 S0 X
Recv: ok
[...]
Send: M569
Recv: X driver mode:	stealthChop
Recv: Y driver mode:	stealthChop`

### Bug Timeline

Not sure. Probably old. Not noticed it before.

### Expected behavior

Expected all axis to turn stealthchop on with the same switch, S1.

### Actual behavior

M569 S0 X turns stealthsop on for X, while M569 S0 Y turns stealthchop off for Y.

### Steps to Reproduce

1. install new 2209 drivers
2. listen to the spreadcycle on X
3. compile and install marlin 2.1
4. find out there is a command to turn stealthchop on. Investigate.
5. run M569 S1 X to turn on stealtchop. Still off.
6. run M569 S0 X to see what happens. Turns on.

### Version of Marlin Firmware

2.0.9.3, 2.0.x-bugfix, 2.1

### Printer model

custom build

### Electronics

SKR 1.4, Biqu H2 v2

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Other things to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.
- [ ] Log output from the host. (`M111 S247` for maximum logging.)
- [ ] Images or videos demonstrating the problem, if it helps to make it clear.
- [ ] A G-Code file that exposes the problem, if not affecting _all_ G-code.

### Additional information & file uploads

In configuration_adv.h i have added #define STEALTHCHOP_Z2 to the list.
There is a STEALTHCHOP_XY.
I also tried STEALTHCHOP_X in addition. No errors, no change.

I have enabled the beta feature SQUARE_WAVE_STEPPING and TMC_DEBUG

I also enabled Z_STEPPER_AUTO_ALIGN

I modified the config from configurations github for biqu b1 abl, as that is what the printer started out as.

Also my drivers are in UART mode.

thisiskeithb, 2022-06-10:
> Other things to include

> A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

You failed to attach your configs.

fuzebox1, 2022-06-10:
[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/8881976/Configurations.zip)

ellensp, 2022-06-11:
you can not just add random defines.

STEALTHCHOP_Z2 and STEALTHCHOP_X are not valid defines. 

Defines enable or disable sections of code
There is no code looking for these defines

ellensp, 2022-06-11:
Provided Serial log is unreadable blob of text.

Cleaned up for readability

```
_All steppers on stealth._ 
Send: M569 
Recv: X driver mode: stealthChop 
Recv: Y driver mode: stealthChop 
Recv: Z driver mode: stealthChop 
Recv: Z2 driver mode: stealthChop 
Recv: E driver mode: stealthChop 
Recv: ok [...] 

_S0 on Y turns Y to spread._ 
Send: M569 S0 Y 
Recv: ok [...] 
Send: M569 
Recv: X driver mode: stealthChop 
Recv: Y driver mode: spreadCycle 
Recv: ok [...] 

_S1 on Y turns stealth back on._ 
Send: M569 S1 Y 
Recv: ok [...] 
Send: M569 
Recv: X driver mode: stealthChop 
Recv: Y driver mode: stealthChop 

_S1 on X turns X to spread. Reversed on X?!_ 
Send: M569 S1 X 
Recv: ok [...] 
Send: M569 
Recv: X driver mode: spreadCycle 
Recv: Y driver mode: stealthChop 

_S0 turns stealth on again. Again, reversed?!_ 
Send: M569 S0 X 
Recv: ok [...] 
Send: M569 
Recv: X driver mode: stealthChop 
Recv: Y driver mode: stealthChop

```

ellensp, 2022-06-17:
I cannot reproduce this issue on marlin bugfix 2.1.x (updated to todays code)

My results form same test as you 

```
M569
21:24:54.044 > X driver mode:   stealthChop
21:24:54.053 > Y driver mode:   stealthChop
21:24:54.062 > Z driver mode:   stealthChop
21:24:54.071 > Z2 driver mode:  stealthChop
21:24:54.080 > E driver mode:   stealthChop
21:24:54.089 > ok

M569 S0 Y
21:25:14.279 > ok

M569
21:25:24.559 > X driver mode:   stealthChop
21:25:24.568 > Y driver mode:   spreadCycle
21:25:24.577 > Z driver mode:   stealthChop
21:25:24.586 > Z2 driver mode:  stealthChop
21:25:24.595 > E driver mode:   stealthChop
21:25:24.604 > ok

M569 S1 Y
21:25:36.651 > ok

M569
21:25:51.943 > X driver mode:   stealthChop
21:25:51.952 > Y driver mode:   stealthChop
21:25:51.961 > Z driver mode:   stealthChop
21:25:51.971 > Z2 driver mode:  stealthChop
21:25:51.980 > E driver mode:   stealthChop
21:25:51.989 > ok

M569 S1 X
21:26:06.036 > ok
M569
21:26:15.104 > X driver mode:   stealthChop    <------------ set to  stealthChop as expected
21:26:15.114 > Y driver mode:   stealthChop
21:26:15.123 > Z driver mode:   stealthChop
21:26:15.132 > Z2 driver mode:  stealthChop
21:26:15.141 > E driver mode:   stealthChop
21:26:15.150 > ok

M569 S0 X
21:26:28.696 > ok
M569
21:26:39.162 > X driver mode:   spreadCycle    <------------ set to  spreadCycle as expected
21:26:39.172 > Y driver mode:   stealthChop
21:26:39.181 > Z driver mode:   stealthChop
21:26:39.190 > Z2 driver mode:  stealthChop
21:26:39.199 > E driver mode:   stealthChop
21:26:39.208 > ok
```

ellensp, 2022-06-17:
S0 Disable stealthChop/Enable spreadCycle   
S1 Enable stealthChop/Disable spreadCycle
Seem to be behaving correctly for X and Y

closed 2022-06-23 by thisiskeithb