MarlinFirmware/Marlin #24323: [FR] Load a Nonflat Base Profile

Mickeylike, 2022-06-10:
### Describe the feature you want

Extension the bed levelling functions, adding the ability to upload a specific nonflat base profile (STL file or synthetic probe matrix) for engraving or CNC maching a nonflat surface such as the inside of a bowl. None of the standard slicing programs seem to permit a nonflat base. However one of the Marlin bed levelling functions apparently generates an STL as a basis. Would it be possible to upload a user defined STL to replace this? Also, for simple adjustments, would it be possible to upload a synthetic probe matrix?

ellensp, 2022-06-11:
That is a slicer function...
https://www.youtube.com/watch?v=VY4cevyGYA4

Mickeylike, 2022-06-11:
Thank-you ellensp. The video was most helpful. It appears to me that all slicers assume a flat base and that uneven surface correction comes from the control board, not the slicer. The Marlin board does have surface mapping functionality and it is apparently possible to run a board driven probe and the board wil subsequently generate an STL to define the Z offset. However, my device is primarily a 3D printer but with a CNC head. There is a magnetic probe bed levelling program for the 3D printing but no physical probe for the CNC. I would like to be able to upload to the board this Z offset map, normally generated by the board, but made to my own specification without the need to physically map a surface.