MarlinFirmware/Marlin #24329: [BUG] Compile error gcode_preview.cpp:241:5: error: 'wait_for_user' was not declared in this scope

jnko, 2022-06-11:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Compiling Marlin fails with error
```
Compiling .pio/build/STM32G0B1RE_btt/src/src/lcd/menu/menu_configuration.cpp.o
Marlin/src/lcd/e3v2/proui/gcode_preview.cpp: In function 'void Preview_DrawFromSD()':
Marlin/src/lcd/e3v2/proui/gcode_preview.cpp:241:5: error: 'wait_for_user' was not declared in this scope
  241 |     wait_for_user = false;
```

Tried both 2.1.x and bugfix-2.1.x

### Bug Timeline

first noticed with 2.1.x

### Expected behavior

Build firmware

### Actual behavior

Failed firmware build

### Steps to Reproduce

1. Extract and write configuration files
2. Build with VSCode / Platform.IO / ABM

### Version of Marlin Firmware

2.1.x / bugfix-2.1.x

### Printer model

Ender 5 Pro

### Electronics

BTT SKR Mini E3 v3.0 (STM32G0B1RE)

### Add-ons

BL-Touch v3.1

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

All changed lines in config are marked with `injk` in comment

[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/8883671/configuration.zip)

ellensp, 2022-06-11:
Provided config files (even with Config files hidden under multiple ./ directories) Compiles fine.

proui is not enabled in provided config files

ellensp, 2022-06-11:
regardless of the configs that do not trigger this issue....

This issue was solved with https://github.com/MarlinFirmware/Marlin/pull/24310  9 hours ago

closed 2022-06-11 by ellensp