MarlinFirmware/Marlin #24330: Failure to compile firmware for Biqu-BX.  invalid conversion from  'const void' to 'FSTR_P'

Antibodyarmy, 2022-06-11:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Bug is consistent after 2.0.9.3 to present. Compiling for biqu bx with the example configs, or with manual configuration will result in this failure to compile. 

Errors are: 
Marduino.h
> conditional expression between distinct pointer types 'const char*' and 'FSTR_P' {aka 'const __FlashStringHelper*'} lacks a cast [-fpermissive] Ln [95, Col 47]
> 
>  marlin/src/HAL/shared/Marduino.h
> ` #define FPSTR(S) (reinterpret_cast<FSTR_P>(S)) `

ui_1024x600.cpp

> invalid conversion from 'const void*' to 'FSTR_P' {aka 'const __FlashStringHelper*'} [-fpermissive] [Ln 736, Col 34]
> operand types are incompatible ("const char *" and "FSTR_P")  [Ln 736, Col 68]
> 
> marlin/src/lcd/tft/ui_1024x600.cpp
>       `FSTR_P const msg = limited ? GET_TEXT(MSG_LCD_SOFT_ENDSTOPS) : FPSTR(NUL_STR);`
> 
>
>
> cannot convert 'const FSTR_P' {aka 'const __FlashStringHelper* const'} to 'const char*' [Ln737, Col19]
>argument of type "FSTR_P" is incompatible with parameter of type "const char *" [Ln737, Col19]
>       `drawMessage(msg);`
[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/8884497/Configurations.zip)

### Expected behavior

I expected a compile to complete, and its hung up on this. 

### Actual behavior

Failure to compile, errors. 

### Steps to Reproduce

copy config examples from biqu/bx to marlin

or

platforimio.ini- 

defauilt_envs = BTT_SKR_SE_BX

configuration.h

#define MOTHERBOARD BOARD_BTT_SKR_SE_BX

#define SERIAL_PORT 4

#define BIQU_BX_TFT70

#define TFT_COLOR_UI

/* Biqu BX Dark Theme */
#define COLOR_BACKGROUND    COLOR_DARK

### Version of Marlin Firmware

latest & bugfix

### Printer model

Biqu BX

### Electronics

Stock - Biqu BX tft 70 and BTT_SKR_SE_BX

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thisiskeithb, 2022-06-11:
https://github.com/MarlinFirmware/Marlin/issues/24228 broke compiling for the BX.

thinkyhead, 2022-06-14:
Should be better now!

closed 2022-06-14 by thinkyhead