MarlinFirmware/Marlin #24337: [BUG] UBL: Z Axis doesn't move during print

radry, 2022-06-12:
### Did you test the latest `bugfix-2.1.x` code?

No, but I will test it now!

### Bug Description

After running UBL to level the bed, storing the mesh and enabling UBL the Z axis doesn't move (compensate) during printing.

### Expected behavior

Z axis moves slightly to compensate the mesh that was created with UBL when printing the first layer.

### Actual behavior

Z axis doesn't move at all.  The Z axis is driven by steppers with threaded rods, so even the slightes movement of the rod should be possible to observe. The mesh has some slight variation, so it should move slightly.

### Steps to Reproduce

1) Have UBL configured in configuration.h
2) G29 P1
3) G29 P3
4) G29 S1
5) G29 A
6) M500

7) Starting GCode in Slicer: 
G28
G29 L1
G29 A

### Version of Marlin Firmware

2.1

### Printer model

Custom

### Electronics

SKR V1.4

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/8886568/Marlin.zip)

Bob-the-Kuhn, 2022-06-12:
I expect it really is compensating, it's just that the movements are too small for you to see.

I think you'll see the same thing if you babystep during the first layer.  Try removing the filament and then putting a piece of paper under the nozzle while printing the first layer.  As you babystep the resistance on the paper will change but you won't notice Z movement.

parallyze, 2022-06-12:
> I expect it really is compensating, it's just that the movements are too small for you to see.

Hmmm.... assuming tr-8x2, 4 threads, 8mm travel per revolution:
1mm z height = 45° rotation. So the movement for 0.1mm is somewhere in the
range of what the hour indicator on a 8mm diameter watch would move in ~8.6 minutes... ^^

> I think you'll see the same thing if you babystep during the first layer. Try removing the filament and then putting a piece of paper under the nozzle while printing the first layer. As you babystep the resistance on the paper will change but you won't notice Z movement.

A bright neon sticky note attached to the rod also greatly helps visualizing tiny movements :)

radry, 2022-06-13:
> I expect it really is compensating, it's just that the movements are too small for you to see.
>
> I think you'll see the same thing if you babystep during the first layer. Try removing the filament [...]

The first layer isn't perfect though. On one side its too close to the bed on the other it's perfect. So it can't be compensating.

thisiskeithb, 2022-06-14:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information:

- Download Marlin [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

Fhalo48, 2022-08-16:
Hi
Any update on this? I'm getting the same problem.

ngsilverwolf, 2022-09-20:
Hi, Having similar issues. not sure if you managed to solve it. I have a log output here. not 100% sure what I'm looking for, unfortunately. Auto Bed leveling is enabled before the print starts off. But half of the print doesn't even touch the base.  I used to have g29 in and then M420 S1 just in case (these were both after g28), But everything seems to be working apart from it actually being level :D  .[](url) 
[log.txt](https://github.com/MarlinFirmware/Marlin/files/9611557/log.txt)

IzzyMan12078, 2022-10-20:
Same issue - in fact, the more I run ABL probe with G29 - the more tilted my mesh gets - yet opposite sides of bed are low/high (compared to mesh) - Appears mesh is applied inverse - but no correcitions are made

closed 2024-09-23 by github-actions[bot]