MarlinFirmware/Marlin #24338: [FR] (Unkown Mainboard with HC32F460 MCU)

n4bstar, 2022-06-12:
### Is your feature request related to a problem? Please describe.

New Ender 2 pro Mainboard with HC32F460 MCU not found in list.

### Are you looking for hardware support?

Yes, try fixing hardware isusses with clean Marlin Firmware for the Creality Ender 2 Pro

### Describe the feature you want

Firmware for New Creality Ender 2 Pro Mainboard.

n4bstar, 2022-06-17:
[Picture from the mainboard](https://github.com/n4bstar/ce2p/blob/main/S1000056.JPG)

EvilGremlin, 2022-06-18:
Bad image, make good sharp hi-res photo of both sides with PCB traces visible and marking on the chip readable.

n4bstar, 2022-06-26:
I have uploaded new images, this is maximum what i can do on quality.

ellensp, 2022-06-26:
Another new CR-FDM this ones a V2.4.S4.???  (last 3 digits are obscured, guessing 170)

n4bstar, 2022-06-26:
> 

yes it 170

th3dstudio, 2022-07-28:
> The Anycubic Kobra and some versions of the Voxelab Aquila also use this chip.
> 
> These machines are delivered with Marlin running on them so someone has already ported Marlin to the HC32F460 and because of the GPL license they are required to publish their modifications. I know that Anycubic has been promising to do just that for a while now but haven't seen any actual code yet.
> 
> @n4bstar you might want to send Creality a message asking for the Marlin source code for your printer.

This is correct. Good luck getting them to comply, I've been barking up that tree for years and just gave up. I hope you get it.

EvilGremlin, 2022-07-28:
This should be same as STM32F469, but i read about differnet interrupt vectors offset and some other funny things
https://translate.google.com/translate?sl=auto&tl=en&hl=ru&u=https://blog.csdn.net/jjp969988159/article/details/114976562&client=webapp

jpraczyk, 2022-08-16:
I've created a dump of the flash from this board. Mine is also 2.4.S4.170. This MCU (HC32f460kcta) has 256 KB of flash and 192 KB of ram available.

[ender2proflash+ram.zip](https://github.com/MarlinFirmware/Marlin/files/9348992/ender2proflash%2Bram.zip)

uski, 2022-09-25:
The Marlin firmware is released under the GPL. I believe the focus of this topic should be to request Creality to release the firmware, as per the GPL forces them to, instead of trying to port the firmware ourselves. This is the whole point of the GPL.

EvilGremlin, 2022-09-25:
Requests are being sent like every day, all pretty much ignored. And let's be honest, their code is mostly garbage in need of complete rewrite.

jpraczyk, 2022-09-25:
> Requests are being sent like every day, all pretty much ignored. And let's be honest, their code is mostly garbage in need of complete rewrite. (mostly because they use very old codebase)

The only public Marlin release for HC32F460 that I've found is https://github.com/Voxelab-64/Aquila_X2/tree/main/firmware/Sources and saying it's hacky is like not saying anything.
So far the creality firmware looks similar to this. Bootloader checks whether firmware update files are stored on the sd (`factory` which also  contains the Bootloader or `user` which only contains Marlin)
It writes the new firmware then jumps to 0x8000. Looks like creality uses the same ddl as Aquila, there is no arduino compatible framework for that mcu so porting will be painful. I may git it a try as soon as my skr mini e3 v3 arrives so that I have something to fall back to, but no promises yet.

EvilGremlin, 2022-09-25:
i'm sure they're just using STM32 framework with custom linker script at most

uski, 2022-09-25:
Part of the issue is that they do this while knowing/assuming they can get away without publishing anything. So far this has proven true.

If they know they are going to have to publish the code, they may make more efforts to write cleaner code and/or just push the changes upstream like so many other companies are doing when they contribute to OSS projects.

Separate discussion in issue #24801 about trying to get them to comply.

jpraczyk, 2022-09-25:
> i'm sure they're just using STM32 framework with custom linker script at most

Highly unlikely, this mcu is not meant to be a drop-in replacement for stm32 similar to gd32.

EvilGremlin, 2022-09-25:
@jpraczyk i linked article above, NVIC address is main difference, might be some other less prominent ones - but using stm32 frameworks is still easiest way
@uski compelled to do cleaner code? LOL, not happening, look at Sovol SV04, look at LVGL_UI history... As were said multiple times before, the only way to make them comply is to damage their revenue enough for them to care.

houseofbugs, 2022-09-26:
> Requests are being sent like every day, all pretty much ignored. And let's be honest, their code is mostly garbage in need of complete rewrite. (mostly because they use very old codebase)

I setup a list and a reporting form today. That printer is now on said list: https://gplviolations.th3dstudio.com/

houseofbugs, 2022-09-26:
> @houseofbugs there is vyper source, they sent it to someone via email [Anycubic Vyper-vyper.zip](https://github.com/MarlinFirmware/Marlin/files/9649544/Anycubic.Vyper-vyper.zip)

Excellent. I will update the list. I dont have time to keep up with it hence why I am going to rely on people reporting things to me.

EvilGremlin, 2022-09-26:
i really should upload some of my stashes as fork branches...

houseofbugs, 2022-09-26:
Maybe someone could make a github repo just to store the sources? Sorted by brand and model in folders. Keep it simple. I would but my time is limited.

EvilGremlin, 2022-09-26:
that too, but proper fork is way more useful, you can throw diff links around

FufuAkuseru, 2022-10-12:
Anycubic has finally released their source code for their [Kobra](https://github.com/ANYCUBIC-3D/Kobra) and [Kobra Max](https://github.com/ANYCUBIC-3D/Kobra_Max) using this MCU. Hopefully somebody more clever than me can put it to good use

lucas19961, 2022-10-13:
anything new ?

InsanityAutomation, 2022-10-13:
We have working source for 2 other machines using the MCU, and they are building it in Kiel, instead of platformIO, so it will not be a simple task to get working in the standard environments. There is at least a way to work though.

houseofbugs, 2022-10-13:
> We have working source for 2 other machines using the MCU, and they are building it in Kiel, instead of platformIO, so it will not be a simple task to get working in the standard environments. There is at least a way to work though.

This is the key here. If someone who knows how to port from Kiel to platformIO can do it for the HC series that would be the major hurdle to getting these supported in Marlin.

InsanityAutomation, 2022-10-13:
> > If someone who knows how to port from Kiel to platformIO
> 
> It sure would be nice for non-Windows users.

Kiel is the arm uVision environment, it's cross platform.

> This is the key here. If someone who knows how to port from Kiel to platformIO can do it for the HC series that would be the major hurdle to getting these supported in Marlin.

PlatformIO needs to add support for the chip first before Marlin can adopt it, but we may be able to work towards that. On the flip side, I'd like to go through this for methods to support other generic cmsis platforms that platformio doesn't support.....

thisiskeithb, 2022-10-13:
> Kiel is the arm uVision environment, it's cross platform.

[If you consider VMs cross-platform, sure](https://developer.arm.com/documentation/ka002387/latest).

InsanityAutomation, 2022-10-13:
> > Kiel is the arm uVision environment, it's cross platform.
> 
> [If you consider VMs cross-platform, sure](https://developer.arm.com/documentation/ka002387/latest).

I was under the impression it ran underneath one of the runtime environments like wine, but it may be a different open source compatible tool that I was thinking of. Been a long time since I've even looked. Not something I commonly need to touch.

InsanityAutomation, 2022-10-13:
Just went and double checked. The uVision ide is windows only. Keil is also part of the arm development studio which is cross platform. Should be no issue opening the project there.

lucas19961, 2022-10-16:
https://www.reddit.com/r/anycubic/comments/y2waxu/tutorial_how_to_build_anycubic_marlin_source_code/ maybe have a look here

doctorevil30564, 2022-11-09:
watching this thread with great interest. I bought a ender 2 pro recently and it has one of the 2.4.S4 mainboards in it. I am having issues with running prints thru octoprint on this printer, where it stutters and it's ruining the quality of the print. the same gcode ran locally runs great. I believe the buffers and baud rate need to be increased in the firmware for the usb serial connection to fix it. I also asked Creality for the firmware and the link they sent me doesn't appear to have a specific repo for this particular chip

jojos38, 2022-11-09:
> e print. the same gcode ran locally runs great. I believe the buffers and baud rate need to be increased in the firmware for the usb

Hey, try to disable power off recovery in the firmware also

mfraser, 2022-12-18:
Disabling power off recovery doesn't help.

doctorevil30564, 2022-12-18:
> > e print. the same gcode ran locally runs great. I believe the buffers and baud rate need to be increased in the firmware for the usb
> 
> Hey, try to disable power off recovery in the firmware also

That didn't work. I wound up buying a used 4.2.7 mainboard and I compiled custom firmware for it using the 2.1.x bugfix marlin code. It's running smoothly now.

mfraser, 2022-12-18:
> That didn't work. I wound up buying a used 4.2.7 mainboard and I compiled custom firmware for it using the 2.1.x bugfix marlin code. It's running smoothly now.

I've got a 4.2.3 board I'm intending to replace it with after Christmas.

mfraser, 2022-12-27:
> I've got a 4.2.3 board I'm intending to replace it with after Christmas.

The 4.2.3 board is now installed. OctoPrint now works.

jpraczyk, 2022-12-30:
If anyone's still running the hc32f460 boards, it looks like they've released the sources for that https://github.com/CrealityOfficial/Ender-2-Pro - MCU_HDSC branch.

ChubbyBunny1, 2023-01-22:
> If anyone's still running the hc32f460 boards, it looks like they've released the sources for that https://github.com/CrealityOfficial/Ender-2-Pro - MCU_HDSC branch.

Yeah it seems they released the code, any clue what to use to build the marlin for the HC32F460 mcu?

EvilGremlin, 2023-01-22:
Keil. uVision IDE or whatever it's called nowadays.

ChubbyBunny1, 2023-01-22:
> 

I have installed the uVision, but there is no .uvprojx file provided by creality. There is only a file called firmware.uvoptx in the workspace folder.

genko, 2023-01-23:
> > 
> 
> I have installed the uVision, but there is no .uvprojx file provided by creality. There is only a file called firmware.uvoptx in the workspace folder.

It seems like they are using the BAT file to compile via CLI.
https://github.com/CrealityOfficial/Ender-2-Pro/blob/MCU_HDSC/workspace/firmware_Release.BAT

ChubbyBunny1, 2023-01-23:
> It seems like they are using the BAT file to compile via CLI. https://github.com/CrealityOfficial/Ender-2-Pro/blob/MCU_HDSC/workspace/firmware_Release.BAT

Correct. I can see in the first line there are some paths. Whare are those for? Of cource the script doesnt run. What compiler should be installed? In the script it has some armcc comands. Is the ARM V5 compiler?

genko, 2023-01-23:
> 
> Correct. I can see in the first line there are some paths. Whare are those for? Of cource the script doesnt run. What compiler should be installed? In the script it has some armcc comands. Is the ARM V5 compiler?

Actually the first line is not needed. It seems to export a lookup path for binaries but looks like they decided to hardcode 
the binary path later.

"C:\Keil_v5\ARM\ARMCC\Bin\ArmCC"

According to the listing files they expect the official arm compiler that ships with uVision in version 5.36. So make sure your compiler binary is available under the path "C:\Keil_v5\ARM\ARMCC\Bin\ArmCC" or do a search and replace for this string with the actual location on your system.

There will be more troubles ahead. You will need files for the architecture as well:

-IC:\Users\DeskCX-1900\AppData\Local\Arm\Packs\ARM\CMSIS\5.8.0\CMSIS\Core\Include
-IC:\Users\DeskCX-1900\AppData\Local\Arm\Packs\ARM\CMSIS\5.8.0\CMSIS\DSP\Include
-IC:\Users\DeskCX-1900\AppData\Local\Arm\Packs\ARM\CMSIS\5.8.0\CMSIS\DSP\PrivateInclude
-IC:\Users\DeskCX-1900\AppData\Local\Arm\Packs\HDSC\HC32F46x\1.0.2\Device\Include

These are include pathes that are hardcoded in all listing files. The first three seem to be from the CMSIS hardware abstraction framework for cortex mcus. You might get this with uVision. Where you get the mcu specific files in the folder HDSC\HC32F46x\1.0.2\Device\Include I currently do not know. Maybe you can check if you can download mcu support packages with uVision for this chip.

ChubbyBunny1, 2023-01-23:
> Actually the first line is not needed. It seems to export a lookup path for binaries but looks like they decided to hardcode the binary path later.
>
> "C:\Keil_v5\ARM\ARMCC\Bin\ArmCC"
>
> According to [...]

Alot of the package information and what is needed is availabe in here https://github.com/CrealityOfficial/Ender-2-Pro/blob/MCU_HDSC/workspace/list/firmware.build_log.htm

I have downloaded what is needed and i am workining towards building the bin file using the .bat. All the .__i filles needed to be opened and configured their paths to match each system. 
Once i manage to build the bin file succesfully without any errors i will make a post, step by step proccess.

ChubbyBunny1, 2023-01-27:
Seems that i am missing a glib.lib file. When i am trying to compile i am getting this error:
C:\Users\Tasos\Downloads\ender 2 pro\Ender-2-Pro-MCU_HDSC\Ender-2-Pro-MCU_HDSC\workspace>"C:\Keil_v5\ARM\ARMCC\Bin\ArmLink" --Via ".\list\firmware.lnp"
Fatal error: L6002U: Could not open file .\glib.lib: No such file or directory
Finished: 0 information, 0 warning, 0 error and 1 fatal error messages.

Anyone have an idea where i can find it? I have searched google but cant find information about the .lib file. Do i need to compile glib somehow?

tom0579, 2023-02-01:
Hi, any progress ?

houseofbugs, 2023-02-01:
> Hi, any progress ?

This question is not helping anything move forward. Please refrain from posting useless questions. You can read the thread above to see what's been going on. Don't be lazy.

tom0579, 2023-02-02:
@houseofbugs : You absolutely right !  I´m  educating myself on this matter in order to be able to contribute with something usefull. The question was more to check if it was some one still working on this.

ChubbyBunny1, 2023-02-02:
I am still working on this, i am waiting for the glib file to be uploaded by creality on their github page.

genko, 2023-02-03:
> I am still working on this, i am waiting for the glib file to be uploaded by creality on their github page.

Hi, I don't think that creality will upload it. It is a open source library. https://gitlab.gnome.org/GNOME/glib
It contains some helper functions that are used to write better C code. So you might need to crosscompile that lib by yourself.

uski, 2023-02-03:
Is the use of glib something specific to this Creality HC32F460 port, or something used elsewhere in Marlin?

If the former, it might be worthwhile checking how exactly glib is being used. If they only use a handful of features, it might be worth moving away from glib, to simplify the build process and align with the rest of Marlin.

We should also make sure it is indeed "the" glib from GNOME, and not something else entirely.

I would suggest removing the dependency from glib, try to build, and see what functions/definitions are missing.

EvilGremlin, 2023-02-03:
i'm sure isn't THE glib

uski, 2023-02-03:
Then I would recommend just removing the dependency from the Makefile, and see what symbols are missing. From there, if code is missing, we can again go push Creality to release whatever source code is missing.

EvilGremlin, 2023-02-03:
my bet it's HMI screen comms part, wont be first time it's being cut from the code release

ChubbyBunny1, 2023-02-04:
> Hi, I don't think that creality will upload it. It is a open source library. https://gitlab.gnome.org/GNOME/glib It contains some helper functions that are used to write better C code. So you might [...]

I havent found any information about the glib.lib file. How do i compile the glib into a .lib file?
I use the Keil uVision V5 ide to compile the code. You need to install the required packages in order for the code to start compiling. Mine compiles for about 20 mins without errors but at the end it throws the glib.lib file missing error.

EvilGremlin, 2023-02-04:
Discalimer: i know nothing about Keil, last time i used it was like 15 years ago to change some defaults in hobby project. 
I suspect it could be proprietary code blob, some form of object file (.o), this way they have secret sauce that don't violate GPL.

ChubbyBunny1, 2023-02-04:
So if i unerstand correctly it is application specific file that is generated by creality? I guess we are hoping for them to upload it. I have already opened an issue 5 day ago...

tom0579, 2023-02-08:
I´ve reach the same point than @ChubbyBunny1 . I try to get rid of the glib.lib entry and got 83 errors regarding undefined symbols.  Without glib.lib believe we cant do nothing. 

.\list\firmware.axf: Error: L6218E: Undefined symbol pinMode(unsigned char, WiringPinMode) (referred from temperature.o).
.\list\firmware.axf: Error: L6218E: Undefined symbol Print::println(unsigned, int) (referred from cardreader.o).
Not enough information to list image symbols.
Not enough information to list load addresses in the image map.
Finished: 2 information, 0 warning and 83 error messages.
".\list\firmware.axf" - 83 Error(s), 0 Warning(s).
Target not created.
Build Time Elapsed:  00:16:38

uski, 2023-02-08:
pinMode is an Arduino function. It should be defined somewhere for the HC32F460 in the code Creality provided. It not, then their release is incomplete. It is very possibly unrelated to the glib.lib missing file, and we can't just guess that it is the only issue with this release.

My recommendation is for someone to first try using the Creality repository to build a "blink LED" firmware. Then upload that to the board using JTAG. This will confirm that we can successfully build and link simple firmware for this board. Then, we can attempt to build Marlin, incrementally.

uski, 2023-02-08:
In the firmware.htm file laying around in the workspace directory of the Creality release, we can see :

pinMode(unsigned char, WiringPinMode) (Thumb, 172 bytes, Stack size 32 bytes, wirish_digital_f1.o(i._Z7pinModeh13WiringPinMode))

Either they made a frankeinstein build sharing bits of HDSC Cortex-M4 libraries, and bits of Arduino STM32F1 libraries, and it works because the cores and peripherals are similar. Or, they modified the STM32F1 Arduino core (unlikely IMO).

What concerns me more is the mention of a "pack" which should be located here:
https://raw.githubusercontent.com/hdscmcu/pack/master/HDSC.HC32F46x.1.0.2.pack

However, I cannot find such file (404). This means that critical include files are potentially missing.

Additionally, the way Creality compiled this firmware is very dirty and unprofessional. We can see in the build log that there is a mention of HAL_SPI.cpp for instance, however, in the firmware release there is only a HAL for STM32F1. So maybe Creality modified the STM32F1 HAL to be used on HC32F460, or maybe they use it as-is, nodoby knows, but gosh their software developpers have horrible coding hygiene and it is definitely not a good advertising for Creality from A to Z (not releasing the source, then releasing a horrible hacked together release, when they could have just done the job properly and mainline it in the first place).

I think I will try to build it with avr-gcc if I find the time, see if I can achieve something

ChubbyBunny1, 2023-02-08:
> In the firmware.htm file laying around in the workspace directory of the Creality release, we can see :
>
> pinMode(unsigned char, WiringPinMode) (Thumb, 172 bytes, Stack size 32 bytes, [...]

What i did is to find the old deleted project file for the uvision on previous releases on their github page. Afther that i installed all the required that you can find in the build log. Indeed the 1.0.2 is not available but from keil uvision you can install the 1.0.7 core pack. Also i opened an issue about a missing petb file and they responded by uploading it to their github. I yet am stuck with the glib.lib file missing

[79 more replies]

closed 2024-01-25 by thisiskeithb