MarlinFirmware/Marlin #24339: [BUG] Marlin crashes when pressing button on BTT SKR TFT35 V3.0 E3 in MARLIN mode (with SPEAKER defined)

autonumous, 2022-06-12:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Marlin crashes and reloads when pressing the LCD button[with SPEAKER enabled in configuration.h].

I've recently tried upgrading my printer to latest firmware version when i noticed the issue.  
Im running an Ender 3 with a BTT SKR Mini E3 V2 mainboard and a BTT TFT35 V3.0 E3 LCD.  

Issue can be replicated with a minimally configured system, eg using current bug fix-2.1.x then making the following changes

configuration.h
```
#define MOTHERBOARD BOARD_BTT_SKR_MINI_E3_V2_0
#define SERIAL_PORT 2
#define SPEAKER
#define CR10_STOCKDISPLAY
```
platform.ini
```
default_envs = STM32F103RE_btt_USB_maple
```

When starting up, all looks ok, the button can be rotated, but when pressed (to access the menu) it crashes

Previously using v2.0.9.3 with no issues.  

Disabling SPEAKER resolved the issue

### Expected behavior

pressing the knob/button, I expect to access the menu

### Actual behavior

pressing the button causes a crash and controller/marlin reloads

### Steps to Reproduce

1. FLash firmware
2. Power on
3. press button

### Version of Marlin Firmware

2.0.9.4, bugfix-2.0.x, 2.1.x, bugfix-2.1.x 

### Printer model

Ender3 Pro

### Electronics

BIGTREE  SKR Mini E3 V2

### Add-ons

BIGTREE TFT35 V3.0 E3 screen

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[bug.zip](https://github.com/MarlinFirmware/Marlin/files/8886875/bug.zip)

Bob-the-Kuhn, 2022-06-12:
Is your BTT TFT35's firmware up to date?

autonumous, 2022-06-12:
> Is your BTT TFT35's firmware up to date?

Yep, V27.    Also tried flashing a newly compiled one yesterday.

thisiskeithb, 2022-06-12:
> default_envs = STM32F103RE_btt_USB_maple

Maple environments are deprecated. Please reflash with the non-maple environment and test again.

autonumous, 2022-06-13:
> > default_envs = STM32F103RE_btt_USB_maple
> 
> Maple environments are deprecated. Please reflash with the non-maple environment and test again.

I think you have identified the issue.     

used STM32F103RE_btt_USB with SPEAKER enabled and better

Arghhh wish I knew that before, thanks.

closed 2022-06-13 by autonumous