MarlinFirmware/Marlin #24342: Temperature calculation for PT1000 not working

KZ-Erge, 2022-06-13:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

In MAX31865.cpp the normalization to 100 ohm (line 464) should be done before line 453. Now if i use pt1000, it always reports 0 degree as line 453 only works for pt100.

### Version of Marlin Firmware

bugfix-2.1.x 

### Printer model

ENDER-5

### Electronics

MKS-SBASE

### Add-ons

MAX31865 board from adafruit, PT1000

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thisiskeithb, 2022-06-13:
>Don't forget to include
> A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

You failed to attach your configs.

zeleps, 2022-06-24:
You're probably right, this was copied and refactored from the original adafruit library without too much thought. I'll try to fix and test it this weekend.

thisiskeithb, 2022-06-26:
Closing since there’s now a PR to fix it: https://github.com/MarlinFirmware/Marlin/pull/24407

closed 2022-06-26 by thisiskeithb