MarlinFirmware/Marlin #24343: [BUG] SD CARD does not start

lipewww, 2022-06-13:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Recently in version 2.1 of Marlin, there was a bug where some LCD's crashed, this happened to me. A solution was discovered, and it fixed my problem, but since then I can't print using the sd card anymore. The problem had to do with the card, but now when trying to start it, nothing happens. Any solution?

### Bug Timeline

Since v2.1

### Version of Marlin Firmware

2.1 bugfix

### Printer model

Anet A6

### Bed Leveling

No Bed Leveling

### Host Software

Cura

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[config marlin 2.1 bugfix.zip](https://github.com/MarlinFirmware/Marlin/files/8893037/config.marlin.2.1.bugfix.zip)

thisiskeithb, 2022-06-13:
This is being worked on: https://github.com/MarlinFirmware/Marlin/pull/24319

closed 2022-06-13 by thisiskeithb

lipewww, 2022-06-13:
> Isso está sendo trabalhado: #24319

where is the solution??