MarlinFirmware/Marlin #24347: [BUG] (Can't print from SD/USB device if ADVANCED_OK is enabled)

freakydude, 2022-06-13:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I print at my stock Artillery Sidewinder X2 with Octoprint (to use it from my network).
Octoprint has some plugins like MeatPack, ArcWelder & BufferBuddy for shrink transmitted data to the printer. 

[BufferBuddy](https://github.com/fflosi/BufferBuddy) tries to keep the Marlin buffers filled. As an requirement, ADVANCED_OK hast to be enabled. It's working fine, if I enable the option. 

But, if ADVANCED_OK is enabled, I can't print direct from SD Card / USB Stick any longer. I can select the gcode file, I can start the print. The display shows the correct target temperatures of nozzel and heat bed. The printing status is at 0%. The printer does nothing. It's not heating at all, not moving. So it stucks here forever.

On the other hand, if I recompile without ADVANCED_OK - printing from SD Card / USB Drive works fine again. But, of course, BufferBuddy stops working.

### Bug Timeline

I don't know

### Expected behavior

The printer prints from SD Card / USB Drive even ADVANCED_OK is enabled

### Actual behavior

If ADVANCED_OK is enabled, I can't print direct from SD Card / USB Stick any longer. I can select the gcode file from the dispaly, I can start the print. The display shows the correct target temperatures of nozzel and heat bed. The printing status is at 0%. The printer does nothing. It's not heating at all, not moving. So it stucks here forever.

### Steps to Reproduce

1. Enable ADVANCED_OK  and recompile Marlin 2.1.0.0
2. Flash a Artillery Sidewinder X2 ( I have the newer GD32F305 version). But maybe it's exactly the same on other printers. Can't test that. 
3. Try to print something from SD Card / USB Stick

### Version of Marlin Firmware

FIRMWARE_NAME:Marlin 2.1 (Jun 13 2022 20:44:10)

### Printer model

Artillery Sidewinder X2

### Electronics

Stock electronics (newer version GD32F305)

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[swx2-marlin-2.1-config.zip](https://github.com/MarlinFirmware/Marlin/files/8893947/swx2-marlin-2.1-config.zip)

Or my github fork.
[Version with ADVANCED_OK](https://github.com/freakydude/Marlin/releases/tag/v2.1.0.0.swx2-fd.01)
[Version without ADVANCED_OK](https://github.com/freakydude/Marlin/releases/tag/v2.1.0.0.swx2-fd.02)
Sorry, here are more changes made ... but to find the root cause, I stripped it down several hours to the ADVANCED_OK option. 

Thank you

InsanityAutomation, 2022-06-13:
Thats an issue with the MKS TFT being used for the USB/SD host. That is also a serial streamer, like octoprint, and doesn't support advanced OK.

closed 2022-06-13 by thisiskeithb

freakydude, 2022-06-14:
Thanks for clarifying. Is there, in theory, an other option to get the buffers managed by serial streamers like octoprint?