MarlinFirmware/Marlin #24348: [BUG] (bug summary) Marlin 2.0.9.4 "Stepping effect" - Test Cube Images observed

VK3SIR, 2022-06-13:
### Did you test the latest `bugfix-2.1.x` code?

No, but I will test it now!

### Bug Description

Version 1.1.9.1 firmware conversion (worked perfectly) to 2.0.9.4.

No matter what I do I cannot resolve this "stepping" effect (see images). It does not occur with v1.1.9.1.

Hardware: Basically a Prusa 3 with inductive level sensor, Heated bed, REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER	(Arduino Mega 2560 based) as BOARD_MKS_GEN_13 .

Image atatched + Configuration files

### Bug Timeline

When started experimenting with 2.0.9.4 LTE

### Expected behavior

Expected print not to skew

### Actual behavior

Print skewed

### Steps to Reproduce

Slice "Cube X Y Z" on both Simplify3d and Prisa Slicer (Latest)

Use Simplify3d or upload to SD to reporduce issues

### Version of Marlin Firmware

2.0.9.4

### Printer model

Cocoon (Prusa 3) With  BOARD_MKS_GEN_13, Inductiove Sensor and Heated Bed

### Electronics

BOARD_MKS_GEN_13

### Add-ons

Inductive bed level sensor

### Bed Leveling

ABL Linear grid

### Your Slicer

Simplify3D

### Host Software

Same as my slicer

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[2.0.9.4.zip](https://github.com/MarlinFirmware/Marlin/files/8894146/2.0.9.4.zip)

VK3SIR, 2022-06-14:
Sorry accidentally closed this. Reopened.

VK3SIR, 2022-06-15:
Attached is an image of the result with v2.1. Note that a version 2.0 LTS version is in the background for comparison.

The issue is FAR WORSe with v2.1. Note that all is fine with the LTE version 1.1.9.1 .

[image]

ellensp, 2022-06-15:
In firmware you have

#define DEFAULT_MAX_ACCELERATION      {300,300,100,100}   

But n your gcode you have 

M201 X600 Y600 Z600 E200 ; sets maximum accelerations, mm/sec^2

The latter is used and may exceed your machines capabilities

VK3SIR, 2022-06-15:
Ellensp – Thanks - I was well aware of that !

If I use a package such as Simplify3D and send it directly from there the results are the same. This file is just one uploaded as a “reference point”. I think I had the settings on earlier tests set so that these figures could be exceeded. Its is a great feature that one in the v2 Codebase 😊

The software, when I examine the code, limits “exceeded values” to the maximum’s anyway. Its painfully slow I know – but gives GREAT prints.

Its not the issue though as it sails through 😊

Something has changed from the v1.1.9.1 LTS to the v2.0.9.4 LTS (and later versions). I am interested to find exactly what has changed that leads to this issue.

Yes I realise that this can be “Mitigated” with Skew settings … But let’s find the root cause and hopefully fix it !

ellensp, 2022-06-15:
everything has changed from v1.1.9.1  to v2.0.9.4
It like comparing an apple seed to a fully grown tree

VK3SIR, 2022-06-15:
😊 … Hence why I am putting this comment out there to find out what we need to do to resolve this.

I should post the v 1.1.9.1 configs at some stage (and will edit this message later when I have more time with these).

Is it a “slicer” config issue?

Is it a “software” issue?

What “philosophical change” change has led to this outcome that I (and I am sure others) are observing?

The last v2 that I worked with was a v2.0.5.x release. The issue was not there then – yet there were extensive “calibration” issues.

ellensp, 2022-06-15:
tested on simulator ( your 2.0.9.4 configs +  #define MOTHERBOARD BOARD_LINUX_RAMPS  
 default_envs = simulator_linux_release) and it thinks it printed fine.
[image]
So issue is specific to real hardware.

ellensp, 2022-06-15:
thing I find odd about your config

#define DEFAULT_AXIS_STEPS_PER_UNIT { 160, 160, 800, 190 }
Are you running 0.9 degree steppers? or 1/8 micro stepping or something else?

#define ENDSTOP_INTERRUPTS_FEATURE
Why is this not enabled?  

#define DISABLE_INACTIVE_EXTRUDER true
should be enabled or disabled, but not set to true

  //#define EEPROM_AUTO_INIT  // Init EEPROM automatically on any errors.
  //#define EEPROM_INIT_NOW   // Init EEPROM on first boot after a new build.
Both of these shouldd really be enabled, to ensure old/bad eeprom setting are not causing issues

#define ST7920_DELAY_1 DELAY_NS(0) 			// Added SIR 16-05-22
#define ST7920_DELAY_2 DELAY_NS(250) 		// Added SIR 16-05-22
#define ST7920_DELAY_3 DELAY_NS(250) 		// Added SIR 16-05-22
These seem very heavy handed...  have you tuned them? are you sure they are even needed?

  #define SD_SPI_SPEED SPI_HALF_SPEED		// #define SD_SPI_SPEED SPI_HALF_SPEED ENABLED SIR 16-06-22
Why? This should not be needed

#define GCODE_MACROS					// #define GCODE_MACROS ENABLED SIR 16-06-22
why? this is consuming 250 (50*5) bytes of ram,    The poor little mega2560 only have 8k, 3 percent used on this alone
Try disabling this.

VK3SIR, 2022-06-15:
Many settings have filtered down from a legacy config based on  v1.1.0b7 (attached below) supplied from the machine's creators. 

[1.1b7.Configuration_adv.h.txt](https://github.com/MarlinFirmware/Marlin/files/8908996/1.1b7.Configuration_adv.h.txt)

[1.1b7.Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/8908998/1.1b7.Configuration.h.txt)

> #define DEFAULT_AXIS_STEPS_PER_UNIT { 160, 160, 800, 190 }
> Are you running 0.9 degree steppers? or 1/8 micro stepping or something else?

The answer here could be "something else" ...

JINSANSHI 42HS34(L)-0954-JA05-D21 as X and Y with 2 x JINSANSHI 42HS34(L)-1204-JA05-D21 steppers in the Z plane . I think that there was a setting for dual motors in the Z plane  that I should perhaps consider ...

It has a different stepper - a 17HS2408 in the extruder (that has undergone many replacements and evolution). 

There are some "issues" noted with the JINSANSHI 42HS34(L)  i.e.:

- http://www.hacking3d.com/2016/05/wanhao-duplicator-i3-v2-stepper-tuning.html

These settings were legacy settings imported from the v1.1.0b7-supplied source. All seemed to work successfully into v2.0.5.3 - the last version that I worked with. 

I will head back to the defaults and start from there - using the references such as the one above as a start point (if required). The guidance here as been much appreciated as it leads me to consider a number of things that I have not considered to this point. I will report back results.

> #define ENDSTOP_INTERRUPTS_FEATURE
> Why is this not enabled? (it should be to take load off of the processor)

The default in the "configuration.h" is to have this disabled. I have left it at default. Thanks for the excellent tip. Perhaps this should be considered to have it enabled in future edits?

> //#define EEPROM_AUTO_INIT // Init EEPROM automatically on any errors.
> //#define EEPROM_INIT_NOW // Init EEPROM on first boot after a new build.
> Both of these should really be enabled, to ensure old/bad eeprom setting are not causing issues

Again the default in the "configuration.h" is to have this disabled. I have left it at default. Thanks for the excellent tip. Perhaps this should also be considered to have it enabled in future edits?

> #define ST7920_DELAY_1 DELAY_NS(0) // Added SIR 16-05-22
> #define ST7920_DELAY_2 DELAY_NS(250) // Added SIR 16-05-22
> #define ST7920_DELAY_3 DELAY_NS(250) // Added SIR 16-05-22
> These seem very heavy handed... have you tuned them? are you sure they are even needed?

and

> #define SD_SPI_SPEED SPI_HALF_SPEED // #define SD_SPI_SPEED SPI_HALF_SPEED ENABLED SIR 16-06-22
> Why? This should not be needed

The device was not reading from the SD card in many versions of the firmware from v1.1.0b7 to v 1.1.9.1, These were obtained via research (I think from https://reprap.org/forum/read.php?415,831502 from past notes that I keep) to overcome issues that I was experiencing. They may not be required for the v2 firmware. This will be tested and I will report back.

> #define GCODE_MACROS // #define GCODE_MACROS ENABLED SIR 16-06-22
> why? this is consuming 250 (50*5) bytes of ram, The poor little mega2560 only has 8k, 3 percent used on this alone
> Try disabling this.

This was an experiment that I was trying that has accidentally flowed across. I will fix and report back. Thanks for this tip and picking it up.

I restate that some of these settings originated from "specimen" 1.1.0b7 firmware provided. There has been CONSIDERABLE evolution since this 1.1.0b7 firmware - and some translations are not just 1:1 for the current LTS releases (though most are). 

It is strange that all is perfect up to and including the 2.0.5.3 release - which was the last time I played with this. I have always tried to do the right thing when I have experienced such issues - report them and hence learn from the creators (such as yourself).

**I cannot thank you enough for your ongoing interest and assistance so far (and ongoing)** All adjustments suggested will be implemented and I will report back results. 

I am dealing with a number of issues on many varying let alone being in the middle of a divorce - so report-backs may take a few days.

VK3SIR, 2022-06-15:
Update.... No success with all changes suggested. 

UBL levelling does not work .... i.e. no levelling cycle is performed. I went back to Bilinear with default settings.

Everything was HALF SIZE when the #define DEFAULT_AXIS_STEPS_PER_UNIT was set to default. 

The print was the "stepped-out" version demonstrated earlier.

Note that there is a "crunch" heard from the X (moving bed) when the Levelling cycle completes and it moves to (0,0).

I am going to try the next suggestion that being that the X, Y and Z steppers are 0.8 degree steppers and do the maths accordingly.

The machine that I am using is variously known as a Wanhao Duplicator i3 V2 (aliases: Monoprice Maker Select and Cocoon Create <== My base machine). I am going to try the example 2.1 script WITH the necessary adjustments for the changed controller.

VK3SIR, 2022-06-24:
Just a minor status update: Back from a short time away; still no success. When I do get the solution - and I will - I will adcd it here.

It seemed to come in from v1.1.9.1 to intermediates to 2.0.9.4 when the facility to "Set the Stepper Driver Type" became mandatory to set.

VK3SIR, 2022-06-26:
This may be the solution: https://3dprinting.stackexchange.com/questions/7950/marlin-2-0-set-stepper-drivers 

The driver is definitely an 8825. I'll report back experiments.

The v 1.1.9.1 firmware is almost useless now; for some reason it is locking up with and "unknown response" or there abouts error.

VK3SIR, 2022-06-26:
Up until v1.1.9.1 (LTS release) there was no need to specify the 8825 driver. 

In the v2.0.9.4 (LTS release) it is critical to ensure that you are using the CORRECT stepper driver as well as the appropriate DEFAULT_AXIS_STEPS_PER_UNIT settings. 

[ To determine this one must (reluctantly) sometimetimes remove the heat sinks over the package ]

Being a Wanhao-based unit it requires:
...
#define X_DRIVER_TYPE  DRV8825
#define Y_DRIVER_TYPE  DRV8825
#define Z_DRIVER_TYPE  DRV8825
...
#define E0_DRIVER_TYPE DRV8825
...
// #define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 400, 500 }
#define DEFAULT_AXIS_STEPS_PER_UNIT {160,160,800,190}

closed 2022-06-26 by VK3SIR