MarlinFirmware/Marlin #24352: [SKR 2.0 resetting automatically] (my skr 2.0 keeps resetting although I tried removing the sd card after booting and stil resets itself and I disabled all the features stated in other fixes that are related to the sd card and usb)

WIDOWMAKER6969, 2022-06-14:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

board resetting itself

### Bug Timeline

new

### Expected behavior

not restart

### Actual behavior

restarts

### Steps to Reproduce

plug in everything and only uncomments the features for moving the axis

### Version of Marlin Firmware

latest

### Printer model

prusa bear

### Electronics

skr 2.0

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

parallyze, 2022-06-14:
Which SKR2 are you using? Rev. A or B, using a F407 or F429? Where's the config files? ^^
What makes you think there's a problem in conjunction with USB and/or SD? Does the machine
also reset if it's doing nothing, maybe around every 180 seconds? Did other versions of Marlin work
before on the same board?

I'm running a SKR 2 Rev B F429, but that one's still on Marlin 2.0.9.3., no spontaneous resets or
anything to watch there....

hcet14, 2022-06-14:
> Which SKR2 are you using? Rev. A or B

In my humble opinion, there is no real difference between Rev. A and B.

@[WIDOWMAKER6969
I'm sure it's no marlin issue. You may move this to ["https://github.com/bigtreetech/SKR-2/issues"](https://github.com/bigtreetech/SKR-2/issues)

hcet14, 2022-06-14:
> What is the difference between the f407 and the f429

> ["https://github.com/bigtreetech/SKR-2/issues"](https://github.com/bigtreetech/SKR-2/issues)

parallyze, 2022-06-14:
> > Which SKR2 are you using? Rev. A or B
> 
> In my humble opinion, there is no real difference between Rev. A and B.

I'm pretty sure there's quite some dead TMC drivers who'd instantly agree on this. Personally I didn't experience
any trouble. When I started using it the F429 environment wasn't available, still no problems. And no dead drivers
so far, although I've swapped them quite a lot (TMC/A4988) when building the printer/setting up Marlin. I doubt
I'll notice any differences like the higher clock frequency on a slow printer like the one it's used in...

> @[WIDOWMAKER6969 I'm sure it's no marlin issue. You may move this to ["https://github.com/bigtreetech/SKR-2/issues"](https://github.com/bigtreetech/SKR-2/issues) I'm running marlin bugfix-2.0.x from 220422 time 16:15 and have no problems, but I'll try the latest bugfix.

The only time I could make the board reset was when messing around with PWM/frequency settings, that made
the board reset every 180 seconds.... hence the curiousity about exact board type/config files :)

@hcet14
Wenn ich das Board mal grillen sollte, gebe ich dir Bescheid. Weiterhin gute Besserung ^^

hcet14, 2022-06-17:
> I'm running marlin bugfix-2.0.x from 220422 time 16:15 and have no problems, but I'll try the latest bugfix.

I tried bugfix-2.1.x from today. No problems.

closed 2022-06-23 by thisiskeithb