MarlinFirmware/Marlin #24353: Biqu BX TFT stuck on marlin splash screen after flashing firmware.

Antibodyarmy, 2022-06-14:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

After compiling the latest bugfix firmware, and flashing firmware.bin the TFT display is stuck on the marlin splash screen. 
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/8896457/Configuration.zip)

[image]

### Expected behavior

finish flashing firmware, and enter the printer TFT printer idle display. 

### Actual behavior

finished flashing firmware, got hung/stuck on marlin splash screen. 

### Steps to Reproduce

compile firmware with biqu/bx config from examples. flash firmware. 

### Version of Marlin Firmware

latest bugfix

### Printer model

Biqu BX

### Electronics

Stock - Biqu BX tft 70 and BTT_SKR_SE_BX

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thisiskeithb, 2022-06-14:
> There are a lot of issues connected to intelligent displays with a µC. Those displays have certainly their own FW. Shouldn't the companies selling those display adapt their FW to new versions of Marlin?

@hcet14: The BX’s TFT is not a standalone TFT. Marlin is fully in charge of controlling it, so this bug is on the Marlin side.

hcet14, 2022-06-14:
> The BX’s TFT is not a standalone TFT.

Sorry, my fault. I understood Antibodyarmy is using a Biqu BX tft 70 ["https://www.biqu.equipment/products/btt-tft43-v3-0-tft50-v3-0-tft70-v3-0-display-touch-screen-two-working-modes"](https://www.biqu.equipment/products/btt-tft43-v3-0-tft50-v3-0-tft70-v3-0-display-touch-screen-two-working-modes).

thisiskeithb, 2022-06-14:
> I understood Antibodyarmy is using a Biqu BX tft 70

They are and I stated above, it is not a standalone TFT. The TFT70 used on the BX is connected to the motherboard via a 40pin cable.

Biqu/BigTreeTech also makes a standalone TFT70 in your link and that is *not* the same one used on the BX. The one in your link connects via two 10-pin IDC + 5-pin serial cables. Those simply external computers / G-code senders.

> Shouldn't the companies selling those display adapt their FW to new versions of Marlin?

Yes, but Biqu/BigTreeTech *does* keep up their TFT firmware (and provides instructions on what to enable in Marlin so they can communicate / work properly), so you should post there instead: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware

At any rate, this is causing unneeded noise and not related to the reported bug.

hcet14, 2022-06-17:
> At any rate, this is causing unneeded noise and not related to the reported bug.

Sorry, I apologize. Thanks for the explanation. I read a little around and think I understood.

thisiskeithb, 2022-06-19:
Thanks to some detective work by @rhapsodyv, this is the breaking commit:

- [🩹 Fix FSTR / PSTR usage](https://github.com/MarlinFirmware/Marlin/commit/4028c1cfc7bcaf78e49bb98b527353ec30a970de)

rhapsodyv, 2022-06-20:
try change the file:
`Marlin/src/HAL/shared/Marduino.h`
code:
`#define FTOP(S) (reinterpret_cast<const char*>(S))`
to
`#define FTOP(S) (reinterpret_cast<const char* const>(S))`
and let me know if it works for you

thisiskeithb, 2022-06-20:
A [stock `bugfix-2.1.x` BX config](https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.1.x/config/examples/BIQU/BX) doesn't work, but a minimal `bugfix-2.1.x` config works:

```
#define MOTHERBOARD BOARD_BTT_SKR_SE_BX
#define SERIAL_PORT 1
#define SERIAL_PORT -1
#define TFT_COLOR_UI
#define BIQU_BX_TFT70
```

rhapsodyv, 2022-06-20:
try disable `SDSUPPORT` to test

thisiskeithb, 2022-06-20:
@Antibodyarmy: Please try the attached firmware: [Biqu-BX-bugfix-2.1.x-firmware.zip](https://github.com/MarlinFirmware/Marlin/files/8942946/Biqu-BX-bugfix-2.1.x-firmware.zip)

It doesn't work on my BX, but works for rhapsodyv & may work for you.

Antibodyarmy, 2022-06-20:
@thisiskeithb It did not work, but now there is a strange behaviour where it boot loops turning on and off

thisiskeithb, 2022-06-20:
> now there is a strange behaviour where it boot loops turning on and off

That's the same thing I see on my setup.

Antibodyarmy, 2022-06-20:
just to clarify my initial issue when I posted, there wasn't a boot loop behavior, but it was stuck on the marlin boot splash screen. the boot loop was introduced with the changes in that firmware you posted and with the changes suggested by rhapsodyv .

thisiskeithb, 2022-06-20:
What about this one? [Biqu-BX-bugfix-2.1.x-firmware-w-24319.zip](https://github.com/MarlinFirmware/Marlin/files/8943328/Biqu-BX-bugfix-2.1.x-firmware-w-24319.zip)

It includes the fix from https://github.com/MarlinFirmware/Marlin/pull/24319

Antibodyarmy, 2022-06-20:
that one is also just stuck on the boot screen, but no on/off loop.

thisiskeithb, 2022-06-20:
@rhapsodyv must have a special / different board, but knowing it doesn't work for you helps. I thought it may have been my board 🙂

Antibodyarmy, 2022-06-20:
I know there are several board/screen-board revisions floating around out there, given the pre release boards(with pi issues) and post Kickstarter changes. My printer is from the later USA Kickstarter rounds.

rhapsodyv, 2022-06-20:
We probably have boards with different bootloader version.

Mine seems flash properly when I use a sdhc (32gb). I only can flash firmware with a 2gb SD. But Keith can use a sdhc to flash his board.

thisiskeithb, 2022-06-20:
I was a beta tester and had an earlier revision board & TFT, but I've been working with rhapsodyv & testing with a "retail" components.

The motherboard I have is an SKR SE-BX 2.0 and TFT is a TFT70-BX V2.2.

thisiskeithb, 2022-06-21:
The features breaking the BX build are `SDSUPPORT` and `TOUCH_SCREEN`. Disable both of those and you'll have a working printer again.

h/t to @rhapsodyv for figuring that out since both of use were using minimal configs & getting different results 🥴

thisiskeithb, 2022-06-21:
More follow-up: Cherry picking changes from https://github.com/MarlinFirmware/Marlin/pull/24319 fixes `SDSUPPORT` on the BX.

This only leaves the broken `TOUCH_SCREEN` support. I've submitted a PR to disable it for now until that can be tracked down: https://github.com/MarlinFirmware/Configurations/pull/757

rhapsodyv, 2022-06-22:
I think I fixed touch bootloop / freeze (build with sd disabled to test touch)
[firmware.bin.zip](https://github.com/MarlinFirmware/Marlin/files/8953504/firmware.bin.zip)

rhapsodyv, 2022-06-22:
Fully working with sd, touch and latest bugfix-2.1.
[firmware.bin.zip](https://github.com/MarlinFirmware/Marlin/files/8953568/firmware.bin.zip)

thisiskeithb, 2022-06-22:
> Fully working with sd, touch and latest bugfix-2.1. [firmware.bin.zip](https://github.com/MarlinFirmware/Marlin/files/8953568/firmware.bin.zip)

Everything works! You're a wizard! 🧙

closed 2022-06-22 by thisiskeithb

Antibodyarmy, 2022-06-24:
[image]

do you guys get a media init fail and no loading of the UI theme after this fix?

thisiskeithb, 2022-06-24:
> do you guys get a media init fail and no loading of the UI theme after this fix?

The photo above is `TFT_COLOR_UI` which looks correct. You will not see the original / square UI from Biqu/BigTreeTech since that was not ported upstream. If you prefer that theme, then you will need to use Biqu/BigTreeTech's BX firmware.

As for the "Media Init Fail", I do not see that. Try reformatting your microSD card.

thisiskeithb, 2022-06-24:
> it will only detect SD cards with the mainboard SD not the screen SD

That is also an expected behavior if you are using the stock BX config from our Configurations repo since it defaults on ONBOARD SD. I would not recommend using the TFT's USB or SD card slots due to possible communication corruption.

> and it cuts in and out with connection on the mainboard

That shouldn't be the case. Are you sure your SD card is fully seated? I started putting my BX back together, so I'll be able to test more this weekend.

thisiskeithb, 2022-06-24:
> the cards are fully seated. If I enter the SD menu, then back out it appears to lose connection with the SD card. If I either re-insert the card or enter/exit the sd menu a few times it will reconnect to the card after spitting the "Media init fail" error.

I never saw that behavior while testing the BX board & TFT on my desk, but I'll see if I can reproduce it once installed in the printer this weekend.

PR #24383 (2022-06-22): Fix BX Touch Freeze
### Description

The touch address was wrong (or it should be configurable?). I took 0x28 from BTT BX repository.

The freeze was happening because it was reading 0xFF from the invalid address and code wasn't handling bad reads.

### Requirements

BX Printer

### Benefits

Fix #24353

### Configurations

Biqu/BX from examples.

### Related Issues

#24353