MarlinFirmware/Marlin #24354: [BUG] SD card access issue. V2.0.9.3 and earlier versions to V2.0.5.4.

Almisuifre, 2022-06-14:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I come from version V2.0.5.4 which no longer compiles for my screen.

So I decided to take a newer version. But the problems keep piling up. I tested all versions since 2.0.5.4. Sometimes I have the screen that bug, sometimes it is the SD card that is no longer reconu. No way to have both!

What I managed to do. A compilation in platformIO.
Enable this:
`//
// RepRapDiscount FULL GRAPHIC Smart Controller
// https://reprap.org/wiki/RepRapDiscount_Full_Graphic_Smart_Controller
//
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER`
...
`/**
 * SD CARD
 *
 * SD Card support is disabled by default. If your controller has an SD slot,
 * you must uncomment the following option or it won't work.
 */
#define SDSUPPORT`
The screen works, the selection button works too. The beep works perfectly. 

### Bug Timeline

I encountered the bug since this morning, when installing LTS version 2.0.9.4, then 2.0.9.3 and earlier

### Expected behavior

Expected behavior: A screen that displays the information and an SD card on which one can read the GCODE files.
Unanticipated behavior: glitch screen, inaccessible SD card

### Actual behavior

But the SD card, no! At the palce it is written:
> Media Unit Fail

### Steps to Reproduce

1. Download marlin in the version you want, including V2.0.9.3.
2. Unzip the zip
3. Open platformIO, search the directory, open the directory.
4. Decommenter:
`
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
#define SDSUPPORT
`
5. Compile
6. Send to map mega 2560
7. Wait for reception.
8. When restarting, the screen reads "Media Init Fail"

### Version of Marlin Firmware

V2.0.9.3

### Printer model

Prusa

### Electronics

Arduino Mega 2560 and Ramps 1.4 EFB, Carte graphique Reprap discount full graphic smart controller

### Add-ons

nothing

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Here are 2 pictures. When the screen works and when the screen fails with some versions. This is not a problem of wire length, bad connection on my part.

Thanks for the help.

[My Config.zip](https://github.com/MarlinFirmware/Marlin/files/8901974/My.Config.zip)

thisiskeithb, 2022-06-14:
This is being worked on https://github.com/MarlinFirmware/Marlin/pull/24319

closed 2022-06-14 by thisiskeithb

Almisuifre, 2022-06-14:
Thank you very much for looking into it. Feel free to ask me to compile for a few days so that I can validate how it works.