MarlinFirmware/Marlin #24355: [BUG] Marlin doesn't work on LERDGE-K on Marlin 2.0.9.3 and up. Display doesn't work on LERDGE-K on Marlin 2.0.8.2-2.0.9.2

odbee, 2022-06-14:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

i have tried getting marlin to work on my LERDGE-K board with no success so far.
Building works in most versions so far.

With versions 2.0.8.2 to 2.0.9.2 i get a reading using the serial port communcation, so marlin is successfully installed, but the display doesn't work.

With versions 2.0.9.3 and upwards the board doesn't work at all after flashing the firmware no matter what I do.

### Steps to Reproduce

set Configuration.h to:

#define MOTHERBOARD BOARD_LERDGE_K
#define SERIAL_PORT 1
#define TFT_GENERIC
#define TFT_INTERFACE_FSMC
#define TFT_RES_480x320
#define TFT_COLOR_UI

as well as 

#define TFT_DRIVER AUTO
or
#define TFT_DRIVER ST7796

### Version of Marlin Firmware

all versions

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/8903217/Config.zip)

Bob-the-Kuhn, 2022-06-17:
Confirmed: (mostly)
2.0.8.1 - LCD and USB work
2.0.8.2 - LCD and USB work - strange, version is 2.0.8.1 in configuration.h, 2.0.8.2 in powerup text on USB
2.0.9.0 - LCD and USB work
2.0.9.1 - LCD and USB work
2.0.9.2 - LCD and USB work 
2.0.9.3 - doesn't proceed beyond **firmware update OK!** boot screen - no LCD, no USB after power cycle
bugfix_2.1.x - doesn't proceed beyond **firmware update OK!** boot screen - no LCD, no USB after power cycle

Bob-the-Kuhn, 2022-06-18:
I see only one 2.0.9.3 commit that mentions Lerdge.

Delete after encrypt. Lerdge encrypt only once (https://github.com/MarlinFirmware/Marlin/commit/b4904cc53e3a8a97fe8047ebe918bc8ea474e120)

odbee, 2022-06-19:
> Confirmed: (mostly) 2.0.8.1 - LCD and USB work 2.0.8.2 - LCD and USB work - strange, version is 2.0.8.1 in configuration.h, 2.0.8.2 in powerup text on USB 2.0.9.0 - LCD and USB work 2.0.9.1 - LCD [...]

could i get your config file to try out the LCD on older versions? or did you use the one i uploaded?

Bob-the-Kuhn, 2022-06-19:
I just made the changes listed in this post.  

I did set TFT_DRIVER to ST7796 in all cases except with 2.0.9.2 I tried both.

odbee, 2022-06-19:
> I just made the changes listed in this post.
> 
> I did set TFT_DRIVER to ST7796 in all cases except with 2.0.9.2 I tried both.

this is very weird. for me it doesn't recognize the display at all.
is it possible for you to send me the whole "Lerdge_K_system" folder you upload to the lerdge board? also, which Version of the display do you have? I have V1.0.2

Bob-the-Kuhn, 2022-06-19:
Which version?

Also, I use VSCode to compile Marlin.

Bob-the-Kuhn, 2022-06-19:
Here's [2.0.9.2 ](https://www.dropbox.com/s/hcj8pe5nflb2d9h/Marlin-2.0.9.2.zip?dl=0).

If you want you can post your config files & I'll compile & post the bin file.

odbee, 2022-06-19:
> Which version?
> 
> Also, I use VSCode to compile Marlin.

i meant the compiled folder you upload to the lerdge board. The "Lerdge_K_system" folder

Bob-the-Kuhn, 2022-06-19:
It's in there.

Bob-the-Kuhn, 2022-06-19:
[Lerdge_K_system.zip](https://github.com/MarlinFirmware/Marlin/files/8936535/Lerdge_K_system.zip)

Bob-the-Kuhn, 2022-06-19:
My LCD has the V1.04 bootloader on it.

rhapsodyv, 2022-06-20:
Check if this make any difference for you: https://github.com/MarlinFirmware/Marlin/issues/24353#issuecomment-1159876546

it seems some of the casts are making the compiler behave crazy in some overload optimizations.

Bob-the-Kuhn, 2022-06-20:
No luck.  Bugfix_2.1.x still hangs after loading the new firmware and is dead after a power cycle.

thisiskeithb, 2022-06-20:
Sounds like the same issues we’re seeing on BX builds.

odbee, 2022-06-20:
> My LCD has the V1.04 bootloader on it.
> 
> V1.0.2 is on the LCD's silkscreen.
> 
> V1.0.1 is on the LerdgeK's silkscreen.

My setup is the same except I have the lerdge V1.0.0.
I have uploaded your firmware with no success on the LCD even though I also have the V1.0.2 LCD.

rhapsodyv, 2022-06-20:
did you try disable `SDSUPPORT`?

Bob-the-Kuhn, 2022-06-20:
Yes, SDSUPPORT is disabled.

rhapsodyv, 2022-06-20:
I have a ledge kit here. I will try use git bisect to find the bad commit.

Bob-the-Kuhn, 2022-06-20:
@rhapsodyv - if you can get me started I can do the git bisect.  

Perhaps these are the steps:

1. download first failing ZIP (2.0.9.3)
2. pull up a command window
3. $ git bisect start
4. $ git bisect bad 
5. $ git bisect good v2.0.9.2

ellensp, 2022-06-20:
you must use git, needs the full comit history
git clone https://github.com/MarlinFirmware/Marlin.git
git bisect start
git bisect good {commit where it still works)
git bisect bad {commit where it is broken}

git will then select a commit half way between these two commits,
You need to add configs and test this and confirm is it good or bad
revert these changes  and tell git the result with one of these

git bisect good or git bisect bad 

git now repeats using a new commit mid way between the new good and bad commits

This is repeated a bunch till it find the one commit that broke it.

Bob-the-Kuhn, 2022-06-20:
Stuck at specifying the good and bad commits.  a185ce22cf6e4fb15250815c5c39318606a7e65a, 2.0.9.2, and V2.0.0.2 all come back as:
```
fatal: Needed a single revision
Bad rev input: 2.0.9.2
```

Bob-the-Kuhn, 2022-06-20:
Finally stumbled through all the git bisect steps.

The offending commit is:
```
commit 6cf6c4cd852bd09c509407057e81e37e36bad52e
Author: Scott Lahteine <github@thinkyhead.com>
Date:   Tue Oct 19 02:56:44 2021 -0500

    🔨 Delete after encrypt. Lerdge encrypt only once
```

Interesting - same label as mentioned previously but different commit number.

rhapsodyv, 2022-06-20:
The only meaningful change I see was the removed the 
`board_build.encrypt = firmware.bin`
try add it back and see if it works.
The addition of the script might be the issue too.

Bob-the-Kuhn, 2022-06-20:
I was able to get 2.0.9.3 to build & run by just backing out the **marlin.py** changes.  

Doing the same change to bugfix_2.1.x did not work.  Usual hang at "firmware update OK" with no USB coms.

What I did on 2.0.9.3:
1. Backed out **stm32f4.ini** changes - hang & no com
2. Backed out **marlin.py** changes - boots & runs
3. Restored  **stm32f4.ini** changes - boots & runs

What I did on bugfix_2.1.x:
1. Backed out **marlin.py** changes - hang & no com
2. Tried the **Marduino.h** FTOP(S) change - hang & no com

Someone needs to verify  **marlin.py** changes is all that was needed on 2.0.9.3.

rhapsodyv, 2022-06-20:
Changing stm34f4.ini + marlin.py didn’t work for 2.1?

Bob-the-Kuhn, 2022-06-21:
bugfix_2.1.x works when both are backed out!

rhapsodyv, 2022-06-21:
Only changing stm32f4.ini isn’t enough?

thisiskeithb, 2022-06-24:
https://github.com/MarlinFirmware/Marlin/pull/24391 has been merged.

closed 2022-06-24 by thisiskeithb

Bob-the-Kuhn, 2022-06-24:
Just tested the latest bugfix-2.1.x.  Compiles with no errors and runs on my Lerdge K.

odbee, 2022-07-03:
> @odbee - does this fix your problems?

hey, sorry for the late reply, it seems like the whole issue occured because i fried my display. the new version works now!. ill be running my printer via octoprint for now, thanks alot!!